Drewsif/PiShrink #133: PiShrink stopped working afer lastest update...

4k3or3et, 2020-04-20:
Hi there

i have noticed that PiShrik was recently updated (several hours ago). When i use the new version i get an error:

`pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
losetup: failed to parse offset: 'YT;
15931539456
4194304
276824064': Invalid argument
tune2fs: No such file or directory while trying to open
Couldn't find valid filesystem superblock.
mount: /tmp/tmp.s8exh6T0Ss: wrong fs type, bad option, bad superblock on , missing codepage or helper program, or other error.
md5sum: /tmp/tmp.s8exh6T0Ss/etc/rc.local: No such file or directory
Creating new /etc/rc.local
mv: cannot stat '/tmp/tmp.s8exh6T0Ss/etc/rc.local': No such file or directory
/media/arturnowok/usb/sh/pishrink.sh: line 207: /tmp/tmp.s8exh6T0Ss/etc/rc.local: No such file or directory
chmod: cannot access '/tmp/tmp.s8exh6T0Ss/etc/rc.local': No such file or directory
umount: /tmp/tmp.s8exh6T0Ss: not mounted.
pishrink.sh: Checking filesystem ...
e2fsck: No such file or directory while trying to open
Possibly non-existent device?
pishrink.sh: Filesystem error detected! ...
pishrink.sh: Trying to recover corrupted filesystem ...
e2fsck 1.44.5 (15-Dec-2018)
e2fsck: No such file or directory while trying to open
Possibly non-existent device?
pishrink.sh: ERROR occured in line 64: Filesystem recoveries failed. Giving up...`

It is sort of confusing becasue both .sh files i have are introducing themselves as version 0.1.2.

When i use the older 0.1.2 version i had before all is going smoothly.

Anybody knows what is going on here?

Thank you guys.

PRochefort, 2020-04-20:
Good morning,
I have the same problem. As a temporary measure, I checked out a previous version with this command after I checked out the master version with git: (ie: from the folder where I cloned the master - I'm no genius with git so there is surely a better way to do this)
git checkout c99b053afb0f7ce2f4416a8c225d467fbb087c5d

This gets me a working version for now.

framps, 2020-04-20:
I just created a PR with the fix. Until it's merged just grab the fix from [my branch](https://github.com/framps/PiShrink/tree/fix133)

4k3or3et, 2020-04-20:
> I just created a PR with a fix. Until it's merged just grab the fix from [my branch](https://github.com/framps/PiShrink/tree/fix133)

Thanks for your work. Your version passes the initial error in my case but it still cannot finish the job in my case. Now i get error:

`pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
pishrink.sh: Checking filesystem ...
/dev/loop0: 117662/269808 files (0.7% non-contiguous), 1007628/1049884 blocks
resize2fs 1.44.5 (15-Dec-2018)
pishrink.sh: Shrinking filesystem ...
resize2fs 1.44.5 (15-Dec-2018)
Resizing the filesystem on /dev/loop0 to 1044391 (4k) blocks.
Begin pass 3 (max = 33)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 9632)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 1044391 (4k) blocks long.

Error: Expecting a partition number.
pishrink.sh: ERROR occured in line 327: parted failed with rc 0`

Would be that kind and tell me what might be wrong in my case?

Thank you.

framps, 2020-04-20:
:flushed: Sorry. My fault. I forgot to add the missing tail which was your initial issue at another place so pishrink failed again. It's fixed now :wink: 

Sorry for the inconvenience :blush:

4k3or3et, 2020-04-20:
> 😳 Sorry. My fault. I forgot to add the missing tail which was your initial issue at another place so pishrink failed again. It's fixed now 😉
> 
> Sorry for the inconvenience 😊

All good now. Thanks for your help.

lurch, 2020-04-20:
I guess this can be closed now?

closed 2020-06-13 by Drewsif