Drewsif/PiShrink #135: Error 247 received from piShrink

johnnypea, 2020-04-20:
Any idea what could cause this?

```
===> Shrinking Image /mnt/remote-dietpi/backups/MyRPi/MyRPi-tar-backup-20200420-142855/MyRPi-backup.dd
pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
losetup: failed to parse offset: 'YT;
64017661952
4194304
276824064': Invalid argument
tune2fs: No such file or directory while trying to open 
Couldn't find valid filesystem superblock.
mount: /tmp/tmp.R6gvzgpoPy: wrong fs type, bad option, bad superblock on , missing codepage or helper program, or other error.
md5sum: /tmp/tmp.R6gvzgpoPy/etc/rc.local: No such file or directory
Creating new /etc/rc.local
mv: cannot stat '/tmp/tmp.R6gvzgpoPy/etc/rc.local': No such file or directory
/usr/local/bin/pishrink.sh: line 207: /tmp/tmp.R6gvzgpoPy/etc/rc.local: No such file or directory
chmod: cannot access '/tmp/tmp.R6gvzgpoPy/etc/rc.local': No such file or directory
umount: /tmp/tmp.R6gvzgpoPy: not mounted.
pishrink.sh: Checking filesystem ...
e2fsck: No such file or directory while trying to open 
Possibly non-existent device?
pishrink.sh: Filesystem error detected! ...
pishrink.sh: Trying to recover corrupted filesystem ...
e2fsck 1.44.5 (15-Dec-2018)
e2fsck: No such file or directory while trying to open 
Possibly non-existent device?
pishrink.sh: ERROR occured in line 64: Filesystem recoveries failed. Giving up...
??? Error 247 received from piShrink
```

johnnypea, 2020-04-20:
I think this could be related to https://github.com/Drewsif/PiShrink/issues/133

framps, 2020-04-20:
> pishrink.sh: Gathering data ...
> losetup: failed to parse offset: 'YT;

Yes, that's a duplicate of #133 and fixed by #134

Drewsif, 2020-04-20:
Closing as duplicate

closed 2020-04-20 by Drewsif