Drewsif/PiShrink #136: Error when zip and debug enabled pishrink.sh -zd

ScherzHerzog, 2020-04-30:
Hi,
at the end of the shrinking process, with debug enabled I get an error in the log. (Marked in **bold** below.

I guess the script is expecting the file still named *.img but only *.img.gz is there

I start pishrink.sh -zd /mnt/usbdisk/ioberry/ioberry-dd-backup-20200430-102936/ioberry-dd-backup-20200430-102936.img

Resizing the filesystem on /dev/loop0 to 927380 (4k) blocks.
Begin pass 2 (max = 243192)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 230)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 18677)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 927380 (4k) blocks long.

Line 320
partnewsize: 3798548480
newpartend: 4071178240
pishrink.sh: Shrinking image ...
Line 345
endresult: 4071178752
pishrink.sh: Using gzip on the shrunk image ...
Line 382
aftersize: 1,2G
pishrink.sh: Shrunk /mnt/usbdisk/ioberry/ioberry-dd-backup-20200430-102936/ioberry-dd-backup-20200430-102936.img.gz from 29G to 1,2G ...
**stat: der Aufruf von stat für '/mnt/usbdisk/ioberry/ioberry-dd-backup-20200430-102936/ioberry-dd-backup-20200430-102936.img' ist nicht möglich: Datei oder Verzeichnis nicht gefunden**

seaniedan, 2020-05-23:
I get a similar error with 

sudo /usr/local/bin/pishrink.sh -vz /tmp/img.img out.img.gz

pishrink.sh: Using gzip on the shrunk image ...
gzip: /home/sean/Documents/4.info/backup/vpn/2020_05_23.img.gz already has .gz suffix -- unchanged
ls: cannot access '/home/sean/Documents/4.info/backup/vpn/2020_05_23.img.gz.gz': No such file or directory
pishrink.sh: Shrunk /home/sean/Documents/4.info/backup/vpn/2020_05_23.img.gz.gz from 30G to  ...
...looks like the script leaves the .gz suffix unchanged in one part of the script, then adds it in another.

framps, 2020-05-23:
@seaniedan Good catch :+1: I fixed this in my branch https://github.com/framps/PiShrink/tree/fix_toolcheck. Please verfiy the fix and I will create a PR.

seaniedan, 2020-05-25:
Yes! That looks like a good fix - perfectly done.

framps, 2020-05-25:
Great. Thank you very much for verifying.

framps, 2020-06-05:
Just an update: I just deleted my fork of pishrink and will no longer create any fixes and PRs.

closed 2020-06-17 by Drewsif