Drewsif/PiShrink #137: ERROR occured in line 131

EDSTOBI, 2020-05-16:
Hi 
Using PiShrink I get the error " is not a file.."
The DD Backup mounts an external drive.
```
sudo mount -t nfs -o soft 10.x.y.z:/volume1/PI4_Backup  /media/pi/3d0c12b1-3924-48e9-becd-815d7b24fb5a/mnt/Diskstation3/backup with success.
BACKUP_PFAD="/media/pi/3d0c12b1-3924-48e9-becd-815d7b24fb5a/mnt/Diskstation3/backup"
BACKUP_NAME="Pi4_Backup"
DD is finishing sudo dd if=/dev/mmcblk0 of=${BACKUP_PFAD}/${BACKUP_NAME}-$(date +%Y-%m-%d).img bs=1MB
sudo /bin/bash /etc/pishrink.sh -d ${BACKUP_PFAD}/${BACKUP_NAME}_$(date +%Y-%m-%d).img ${BACKUP_PFAD}/${BACKUP_NAME}_$(date +%Y-%m-%d)-small.img
```
Dose PiSchrink have a problem wit the BACKUP_PFAD to the file?
Txh

framps, 2020-05-16:
We need more information :wink: 

Invoke pishrink the following way:
```
script pishrink.log
sudo /bin/bash -e /etc/pishrink.sh -d ${BACKUP_PFAD}/${BACKUP_NAME}_$(date +%Y-%m-%d).img ${BACKUP_PFAD}/${BACKUP_NAME}_$(date +%Y-%m-%d)-small.img
exit
```
and show the contents of pishrink.log.

EDSTOBI, 2020-05-18:
Hi
What I found out
my backup script is not pathing over the variables 
```
BACKUP_PFAD="/media/pi/3d0c12b1-3924-48e9-becd-815d7b24fb5a/mnt/Diskstation3/backup"
BACKUP_NAME="Pi4_Backup".
```
Bei writing the full name it works.
`sudo /bin/bash -e /etc/pishrink.sh -d /media/pi/3d0c12b1-3924-48e9-becd-815d7b24fb5a/mnt/Diskstation3/backup/Pi4_Backup_$(date +%Y-%m-%d).img /media/pi/3d0c12b1-3924-48e9-becd-815d7b24fb5a/mnt/Diskstation3/backup/Pi4_Backup_$(date +%Y-%m-%d)-small.img`

Until next different error, seel log

> pishrink.sh v0.1.2
pishrink.sh: Copying /media/pi/3d0c12b1-3924-48e9-becd-815d7b24fb5a/mnt/Diskstation3/backup/Pi4_Backup-2020-05-17.img to /media/pi/3d0c12b1-3924-48e9-becd-815d7b24fb5a/mnt/Diskstation3/backup/Pi4_Backup-2020-05-17-small.img... ...
pishrink.sh: Gathering data ...
Line 195
beforesize: 7,5G
parted_output: BYT;
/media/pi/3d0c12b1-3924-48e9-becd-815d7b24fb5a/mnt/Diskstation3/backup/Pi4_Backup-2020-05-17-small.img:7948206080B:file:512:512:msdos::;
1:4194304B:272630271B:268435968B:fat32::lba;
2:276824064B:7948206079B:7671382016B:ext4::;
partnum: 2
partstart: 276824064
tune2fs_output: tune2fs 1.44.5 (15-Dec-2018)
Filesystem volume name:   rootfs
Last mounted on:          /
Filesystem UUID:          26992cbf-74a1-4b6d-84eb-802b70f9fd9c
Filesystem magic number:  0xEF53
Filesystem revision #:    1 (dynamic)
Filesystem features:      has_journal ext_attr resize_inode dir_index filetype needs_recovery extent flex_bg sparse_super large_file dir_nlink extra_isize
Filesystem flags:         unsigned_directory_hash 
Default mount options:    user_xattr acl
Filesystem state:         clean
Errors behavior:          Continue
Filesystem OS type:       Linux
Inode count:              459360
Block count:              1872896
Reserved block count:     82225
Free blocks:              644514
Free inodes:              293050
First block:              0
Block size:               4096
Fragment size:            4096
Reserved GDT blocks:      208
Blocks per group:         32768
Fragments per group:      32768
Inodes per group:         7920
Inode blocks per group:   495
Flex block group size:    16
Filesystem created:       Thu Jun 20 19:05:22 2019
Last mount time:          Fri May 15 00:30:53 2020
Last write time:          Thu Jun 20 19:06:13 2019
Mount count:              85
Maximum mount count:      -1
Last checked:             Thu Jun 20 19:05:22 2019
Check interval:           0 (<none>)
Lifetime writes:          81 GB
Reserved blocks uid:      0 (user root)
Reserved blocks gid:      0 (group root)
First inode:              11
Inode size:	          256
Required extra isize:     32
Desired extra isize:      32
Journal inode:            8
First orphan inode:       56814
Default directory hash:   half_md4
Directory Hash Seed:      8ffe9782-ad3f-424f-9a01-375205312664
Journal backup:           inode blocks
currentsize: 1872896
blocksize: 4096
Creating new /etc/rc.local
pishrink.sh: Checking filesystem ...
rootfs: Inodes that were part of a corrupted orphan linked list found. 
rootfs: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
	(i.e., without -a or -p options)
>

framps, 2020-05-18:
> BACKUP_PFAD="/media/pi/3d0c12b1-3924-48e9-becd-815d7b24fb5a/mnt/Diskstation3/backup"
> BACKUP_NAME="Pi4_Backup".

Try
```
export BACKUP_PFAD="/media/pi/3d0c12b1-3924-48e9-becd-815d7b24fb5a/mnt/Diskstation3/backup"
export BACKUP_NAME="Pi4_Backup"
```

EDSTOBI, 2020-05-20:
Hi
Script is running without problems over the last days.
THX

closed 2020-05-20 by EDSTOBI