Drewsif/PiShrink #138: Pishrink fails with error in line 64

andohuman, 2020-05-22:
For context I have Ubuntu 20.04 on my pi sd card. I made an image using dd. I'm trying to shrink that image using this pishrink. I previously tried to shrink an archarm install, which failed and I assumed it was because arch wasn't supported.

[image]

I couldn't find any other issue similar to mine. Any help would be greatly appreciated.

Computer information :- Alienware 15 R2 running Pop os 20.04. (but previously the script also failed in a manjaro and Ubuntu 20.04 too)

kaoscoach, 2020-05-31:
I get this same error when trying to shrink Kali. I found that if I immediately re-run the command, it works.

andohuman, 2020-05-31:
Does it work when you actually boot your pi? Then it's just some weird bug. I think I've never tried running it twice. I conceded and installed raspbian. Thank you.

kaoscoach, 2020-05-31:
It does work. I am not sure why pishrink fails the first time. My first guess is that it has something do with the creation or using /dev/loopX but I am not sure. In line 201 it mounts $loopback but I think this command is not working as expected. It can't find the newly created rc.local file that it appears to create, so I think something is wrong with the mount. Trying to do the script commands manually, but haven't found the problem, yet.

framps, 2020-05-31:
Could you recreate the issue and capture the run log by executing following commands?

```
script pishrink.log
sudo pishrink.sh ... # pass your command line options to shrink your image
exit
```
Then attach `pishrink.log`here for further analysis.

andohuman, 2020-05-31:
Hi @framps , I've attached the log file for your reference. For context, I tried shrinking an image containing ubuntu 20.04.
 
[pishrink.log](https://github.com/Drewsif/PiShrink/files/4708035/pishrink.log)

andohuman, 2020-05-31:
@kaoscoach So I just tried running the script twice as you said and it does not work in my case. I'm still getting the same error.

kaoscoach, 2020-05-31:
File attached.
[pishrink.log](https://github.com/Drewsif/PiShrink/files/4708175/pishrink.log)

kaoscoach, 2020-05-31:
I hope it is implied, but just to make note with the log I just attached. It doesn't work the first time I run the script, but works the 2nd time.

framps, 2020-05-31:
Thank you very much for the logs. I think I located the root cause. Would you please check whether my fix in my [branch](https://github.com/framps/PiShrink/tree/issue138) works now. If not please provide the run log.

kaoscoach, 2020-05-31:
[jrather@4fwzdb2 pi-kali]$ sudo ../pishrink.sh pi-kali-20200531.img 
pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
../pishrink.sh: line 269: syntax error near unexpected token `else'
../pishrink.sh: line 269: `else'

kaoscoach, 2020-05-31:
If I add the command,
partprobe "$loopback"
prior to line 201 which is "mount "$loopback" "$mountdir"
The script works the first time.

kaoscoach, 2020-05-31:
Sadly, I do not know how to do this git stuff.

framps, 2020-05-31:
> ../pishrink.sh: line 269: syntax error near unexpected token `else' ../pishrink.sh: line 269: `else'

My fault :cry: 

I fixed it. Please try again.

andohuman, 2020-06-01:
@framps Thank you! The script from your branch worked without any issue, shrinking a 32Gb image to 5.7 GB. And just for my own satisfaction, I further shrank it down to 1.8Gb using xz. 

Do you think this script will also work for non-pi images? It'd be great if it worked for a normal 64bit install but I'm specifically looking to use this script with jetson nano images. Let me know your thoughts and meanwhile, you can close this issue. Thanks again!

framps, 2020-06-01:
Great it works for you. Thank you very much for your verification.

@kaoscoach : Would you please also verify my fix? In contrast to @andohuman your filesystem state was not clean but should be cleaned now. If you confirm the fix works for you too I'm going to create a PR for pishrink.

framps, 2020-06-01:
> Do you think this script will also work for non-pi images? It'd be great if it worked for a normal 64bit install but I'm specifically looking to use this script with jetson nano images. Let me know your thoughts and meanwhile, you can close this issue. Thanks again!

I think Drewsif has some plans on this. @Drewsif  Please correct me if I'm wrong.

kaoscoach, 2020-06-01:
My install of Kali does not include the /etc/rc.local file. So checking for it's existence isn't going to work on my image. Verifying the losetup command by adding the partprobe command before the mount command however does fix my issue.

For edification, I installed Kali using the "Kali Linux RaspberryPi 2 (v1.2), 3 and 4 (64-Bit)" kali-linux-2020.2a-rpi3-nexmon-64.img.xz image from https://www.offensive-security.com/kali-linux-arm-images/ I am using a Pi 4. It is my assumption that PiShrink will essentially work on any image with two partitions as long as the 2nd partition is the root filesystem and it is formatted with ext(2,3,4).

framps, 2020-06-01:
> So checking for it's existence isn't going to work on my image.

That's what I [changed](https://github.com/framps/PiShrink/commit/76826a13f70540dfc31616580f8c5a232fd03995): Now it's checked whether the /etc/rc.local exists and if not no md5sum is executed which caused the issue. Now the loop device will not be umounted any more and no parted is required. Would you please check my fix?

kaoscoach, 2020-06-01:
I did run your fix and it does work. I realize that my install of Kali does not include /etc/rc.local but I see that file is created for the resize on boot. My guess is that @andohuman Ubuntu 20.04 does not have /etc/rc.local either. Maybe partprobe isn't needed, even though it probably doesn't hurt either.

framps, 2020-06-01:
> I did run your fix and it does work.

Great. Thank you very much for the fix verification.

> I realize that my install of Kali does not include /etc/rc.local but I see that file is created for the resize on boot.

/etc/rc.local is used to resize the image. If there exists a previouse /etc/rc.local this file will put in place after resize. Otherwise /etc/rc.local used for resizing is deleted after reboot.

> Maybe partprobe isn't needed, even though it probably doesn't hurt either.

Agree. That's why I didn't use it in my fix.

I will create a PR now.

andohuman, 2020-06-03:
Sorry for reopening the issue but I really don't know where to post this. Pishrink actually works on jetson images. The reason it didn't work earlier was because I was trying to shrink the image without doing the first boot. I've been use jetson for 2 days setting up things and thought I'd just give it a try and the script actually compressed a 32G image to 15G. 

[image]

framps, 2020-06-05:
Just an update: I just deleted my fork of pishrink and will no longer create any fixes and PRs.

Drewsif, 2020-06-13:
Glad to hear it works on other images!

@framps thanks for all the help, your contributions were very much appreciated <3

closed 2020-06-13 by Drewsif