Drewsif/PiShrink #141: Add an option to reduce size without important files losses

xDaKillerx, 2020-05-28:
Hi. So I have an idea . I think it'd be nice to make another option to clean logs, and useless stuff before resizing and compressing the image. That way it'll be smaller than it used to. Also idk if my logic is right or not but as far as I know. The script still keep the swap file right? Now, can we remove the file before resizing and compressing? Does the device on the boot after it's flashed will automatically make the swap file again? Or no? If not. Maybe we could make it so that it'll check the size for swap file and then zero it. So when it's compressed. It's easier and smaller. What do you think? Thanks!

framps, 2020-05-28:
In general I think that's good idea. But I frankly think pishrink should follow the [Unix philosophy](https://de.wikipedia.org/wiki/Unix-Philosophie) 
```
Make each program do one thing well. To do a new job, build afresh rather than complicate old programs by adding new "features".
```

xDaKillerx, 2020-05-28:
Hmm I see. Does compressing the compressed image (xz on both) will yield better compression ratio ??

framps, 2020-05-28:
Looks like you want to shrink your image as small as possible. Frankly I don't understand this because nowadays disk space is quite cheap. But anyhow - no - I have no idea whether you will get any better compression rates because I'm fine with the standard non compressed shrinked image.

xDaKillerx, 2020-05-28:
Okay then. Thanks for the info!! Cheers.....

closed 2020-05-28 by xDaKillerx

Drewsif, 2020-05-28:
Its not a bad idea but Im also trying to keep it compatible with most OS's so I wouldn't want to delete or zero files that may cause issues. I do like the idea but I don't really see it in the future.