Drewsif/PiShrink #143: PiShrink a Persistent Ubuntu Live USB

unclehowell, 2020-06-01:
I made a Ubuntu Persistent Live USB which I want to make available to my colleagues and developers so they have the latest correct environment, to maintain and develop the Pi software I've built and the business operation. Can PiShrink be used for this and if so how?

Drewsif, 2020-06-02:
Are you trying to shrink the USB image? It may work if your last partition is the root partition but I've not tested it.

ch1x0r, 2020-06-02:
Also, you can truncate the image. 
David Sugar suggested I port respin to Pi /ARM. I guess there is a need now.
I just dd after truncating.

unclehowell, 2020-06-03:
I did dd, various methods, img.gz is still 11GB. I know the proper method is to make it build the persistent live usb from binaries, which will make the image tiny. Like with ua-netinst, which gives me a self-building version of the latest Debian in a 60Mb image. But I'm just curious at this stage how efficient I can make a backup of the Ubuntu Live USB I have. I read somewhere that I must use gparted to manually resize each partition down to the size of the content then follow the processes in the Pishrink code.  I'll mess with it soon as I'm free next and report back. PiShrink and ua-netinst for Ubuntu Persistent Live would be epic, I'm suprised it doesn't already exist. Hope this whole step gets easier and better.

ch1x0r, 2020-06-03:
Yes. Use gparted to truncate. Also, I'm the lead dev for Libre Respin - Debian version and when tools exist already, I don't bother... but if you see a need - for arm, and others have stated... there is one, then I will consider porting. Let me know.

Drewsif, 2020-06-17:
Yeah since PiShrink only resizes the last partition the others you would have to do manually. When building the Live USB I suggest you manually create the partitions and have a `/boot` partition as the first partition then allocate the rest of the space ext4 for `/`.

Also if you compress your image with `xz` you will notice better compression ratios. It is slower to compress but worth it in my opinion.

closed 2020-06-17 by Drewsif