Drewsif/PiShrink #145: PiShrink failing with "The superblock could not be read"

lynnejohn, 2020-06-11:
- Backing up a Ubuntu server 19.01 with a Mate Desktop(Pi4 4Gb)
- Made the initial backup .img file while running on a different Pi4. I used dd from the USB to SD card adapter to a externally connected hard drive(USB to sata.)

Using the uncompressed image file I have taken the backup to a new sd card and it boots and runs fine.

When I run PiShrink.sh I get the following error:
****************************************************************************************
pishrink.sh v0.1.2
pishrink.sh: Copying /media/tester/5a119587/home/tester/resoil_backup.img to /media/tester/5a119587/home/tester/smaller.img... ...
pishrink.sh: Gathering data ...
md5sum: /tmp/tmp.nwBpiPpvD2/etc/rc.local: No such file or directory
Creating new /etc/rc.local
mv: cannot stat '/tmp/tmp.nwBpiPpvD2/etc/rc.local': No such file or directory
pishrink.sh: Checking filesystem ...
e2fsck: Invalid argument while trying to open /dev/loop5
/dev/loop5: 
The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem.  If the device is valid and it really contains an ext2/ext3/ext4
filesystem (and not swap or ufs or something else), then the superblock
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>
*******************************************************************************************

     How do I obtain more information on this error to help resolve what I'm doing wrong?

Thanks

lynnejohn, 2020-06-11:
I attached the log and everything looks good until it hits the md5sum.

[pishrink.log](https://github.com/Drewsif/PiShrink/files/4767366/pishrink.log)

lynnejohn, 2020-06-17:
I've taken the original sdcard and run e2fsck on the Linux partition "sudo e2fsck -f -y -v -C 0 '/dev/sda2'":
e2fsck 1.44.5 (15-Dec-2018)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure                                           
Pass 3: Checking directory connectivity                                        
Pass 4: Checking reference counts
Pass 5: Checking group summary information                                     
                                                                               
      431883 inodes used (23.06%, out of 1873248)
         880 non-contiguous files (0.2%)
         371 non-contiguous directories (0.1%)
             # of inodes with ind/dind/tind blocks: 0/0/0
             Extent depth histogram: 389660/216
     2155158 blocks used (27.81%, out of 7749376)
           0 bad blocks
           1 large file

      312492 regular files
       66667 directories
           9 character device files
           0 block device files
           0 fifos
         162 links
       52693 symbolic links (41977 fast symbolic links)
          13 sockets
------------
      432036 files

I then took that sdcard, made a .img and tried pishrink on it and got the same failure:
sudo dd if=/dev/sda of=/media/pi/7209be/home/tester/bkup_16_June_2020.img bs=1024 status=progress
sudo pishrink.sh -d /media/pi/7209be/home/tester/bkup_16_June_2020.img /media/pi/7209be/home/tester/smaller.img

I don't see anything blatantly wrong with the Linux partition. Could pishrink be unhappy about some other part of the image?

Drewsif, 2020-06-17:
Have you tried the latest version? This looks like a problem that may have been fixed (hopefully)

lynnejohn, 2020-06-17:
I completely failed to see a new version that says it was changed "2 days ago"(can it be more obvious). My original run showed a version of v0.1.2 and when I manually checked the script via the web page, I saw the same version.  It is now happily crushing down our Ubuntu Mate test station image. 

Thank you very much for all the effort you put in on this project, I can't tell you how much time you saved me in creating a reliable back up methodology for our test stations. If you need Pi hardware to test on or take donations please let me know.  

Thanks again.

closed 2020-06-17 by lynnejohn