Drewsif/PiShrink #147: Xz failed on latest pishrink

White7561, 2020-06-14:
pishrink.sh v0.1.2                                                         
pishrink.sh: Gathering data ...                                            
Creating new /etc/rc.local                                                 
pishrink.sh: Checking filesystem ...                                       
rootfs: 150907/1942592 files (1.1% non-contiguous), 1718091/7799680 blocks 
resize2fs 1.44.5 (15-Dec-2018)                                             
pishrink.sh: Shrinking filesystem ...                                      
resize2fs 1.44.5 (15-Dec-2018)                                             
Resizing the filesystem on /dev/loop0 to 2063013 (4k) blocks.              
Begin pass 2 (max = 82713)                                                 
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX     
Begin pass 3 (max = 239)                                                   
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX     
Begin pass 4 (max = 14243)                                                 
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX     
The filesystem on /dev/loop0 is now 2063013 (4k) blocks long.              
                                                                           
pishrink.sh: Shrinking image ...                                           
pishrink.sh: Using xz on the shrunk image ...                              
xz:  -v: Invalid multiplier suffix                                         
xz: Valid suffixes are `KiB' (2^10), `MiB' (2^20), and `GiB' (2^30).
pishrink.sh: ERROR occured in line 405: xz failed with rc 0

My command to run PiShrink : sudo ./pishrink.sh -Zarv Primary.img

Drewsif, 2020-06-14:
Thanks for reporting the issue :)

closed 2020-06-14 by Drewsif

xDaKillerx, 2020-06-15:
latest script now works. but the verbose on xz is no longer working. hmm . why wasnt it working before? it used to work just fine? ( the verbose flag on xz )

framps, 2020-06-15:
In https://github.com/Drewsif/PiShrink/commit/370e88d58194c0c84f4b7cd962a8e2fdc79c602c this  bug was introduced when quotes were added in lines 395 and 403. `-v` now doesn't work any more because @Drewsif deleted In https://github.com/Drewsif/PiShrink/commit/66355825bd774b083057b13f6f9bae068ee53223 line 398 which turned on verbose mode if requested with option `-v`.

xDaKillerx, 2020-06-15:
ahh i see. do you think its gonna come back or not? thankss

framps, 2020-06-15:
It will take me a couple of minutes to create a PR for this but I don't create any longer PRs for pishrink.

xDaKillerx, 2020-06-15:
ah. ok then thanks for the help :)

Drewsif, 2020-06-15:
Did you actually use the verbose output of xz? I assumed it was not useful to most people. I can re-add it if it is useful

framps, 2020-06-15:
Option `-v` was disabled also for gzip when you deleted line 398 and now there is no option `-v` supported any more by pishrink. 

If you think option `-v` it's useless the option should have been  removed from getopts options and the script help and README should have been updated accordingly :wink:

White7561, 2020-06-15:
> Did you actually use the verbose output of xz? I assumed it was not useful to most people. I can re-add it if it is useful

If you have time please re-add it. I always use it so I know the progress and the ETA. thanks!!

Drewsif, 2020-06-15:
Added back in, sorry about removing something that was used. I never realized that xz gave you progress on verbose so I'm definitely going to start using that. I was testing with gzip and it's verbose output is not anywhere near as useful

White7561, 2020-06-16:
No no no its fine :) . Thanks for adding it back!! . Yeah the verbose is actually very useful on xz . If you add the verbose flag again on xz you can see the compression settings and how much ram it'll use. Pretty useful!!

commit 66355825 (2020-06-14): Remove verbose option setting from zipping tool options

Closes #147