Drewsif/PiShrink #148: PiShrink works!

wschwack, 2020-06-25:
That's a great tool, thank you very much!

yuriw, 2020-06-28:
I have not actually tested how to create a bootable SD card with shrunk image, but yes it seems working fine so far, thank you !

neilyoung, 2020-06-28:
It works like a charm. It should be noted somewhere, that it is a MUST to add parameter `-s` in order to prevent filesystem expansion on first start (buster), if the shrank image contains a read-only filesystem.

yuriw, 2020-06-28:
@neilyoung what exactly `-s`  do ?

neilyoung, 2020-06-28:
-s         Don't expand filesystem when image is booted the first time

yuriw, 2020-06-28:
I created an img file with and without `-s` option and the final file size is very similar
I am getting new sd cards and test it soon.

But what the difference would I expect to see using two those image files on the resulting sd card?

neilyoung, 2020-06-28:
The difference is not in size, it is in behaviour. If you backup an SD which has a RO filesystem, shrink it and restore again, it will not work (if it is Raspberry OS Buster), since the inbuilt filesystem expansion finds required directories and files not writable. The PI will not successfully boot from this SD anymore.

This might also happen, if you restore from the original, not shrank image, but with the original image there is no chance to prevent the implicit file system expansion of Buster. PIShrink came with this `-s` switch, offering me to skip the FS expansion and - tadaaa - this was the exit and solution. Letting PIShrink care for the suppression of the unnecessary, unwanted and impossible expansion of the RO filesystem made the restored image bootable again.

During the attempts to boot the failing restored RO system logged to `journalctl`. For a short time it was possible to SSH, until the system rebooted and was trapped again. This gave me the chance to come behind this.

Here the error traced, this gave me a clue, what was wrong:

```
Jun 25 19:01:00 dfstreamer rc.local[590]: rm: cannot remove '/etc/rc.local': Read-only file system
Jun 25 19:01:00 dfstreamer rc.local[590]: cp: cannot remove '/etc/rc.local': Read-only file system
Jun 25 19:01:00 dfstreamer rc.local[590]: Partition #2 contains a ext4 signature.
Jun 25 19:01:00 dfstreamer rc.local[590]: /usr/bin/raspi-config: 182: /usr/bin/raspi-config: cannot create /etc/init.d/resize2fs_once: Read-only file system
```
During this time the green LED on a PI3A+ blinks in a certain rhythm, which is unusual.

So `-s` plays no role with the restore of an RW filesystem, but enables the boot from a restored RO filesystem. 

HTH

yuriw, 2020-06-28:
I can’t claim that all what you said I understood, but it sounds like for R Pi SD card copy I better use `-s` switch 

Thx 👍

PS: wonder if there’s any plan to enable pishrink to be able to restore img …

neilyoung, 2020-06-28:
Well, it depends. Say your image is coming from a 16 GB SD and has been shrank to 3.5 GB (my case). If you have shrank with -s that would mean, that on your restored SD, which might be 64 GB big, only the 3.5 GB are usable. I don't think this makes sense on a RW system, since the biggest chunk of the SD is unusable, which can cause problems in future. Let the system expand the filesystem on first boot, this only slows down the first boot a bit.

As said, -s is only vital for RO systems. An RO system doesn't need to be expanded, because all RO parts are "ROM" so to say (the firmware) and all writable filesystems are on RAM disk.

neilyoung, 2020-06-28:
> PS: wonder if there’s any plan to enable pishrink to be able to restore img …

This is not neccessary, IMHO. If you are on Ubuntu, the restore is more or less just one command at terminal. You not even need to have an app for that :)

yuriw, 2020-06-28:
Well that’s what I’m not 100% about - RO vs RW fs

All what need ATM be able to make a bootable copy of my SD card used to boot and run my R Pi 4, would you say use `-s` for this use case ?

On the item, I guess you mean `dd` and that’s true, still it be good to have  `one stop` solution 🤔

neilyoung, 2020-06-28:
Yes, `dd` is the way to go. Reliable and good :)

For a PI in "rough" environment (e.g. unbuffered power supply) it is vital to operate from an RO filesystem, since if you hit an SD in the middle of a write with a power cycle, you might most likely have to say Good Bye to the SD, so your system will not boot anymore.

That's why: Don't go w/o RO :)

neilyoung, 2020-06-28:
> All what I need ATM be able to make a bootable copy of my SD card used to boot and run my R Pi 4, would you say use -s for this use case ?

Sorry, did oversee your question. Go w/o -s.

yuriw, 2020-06-29:
@neilyoung So far I am out of luck.
Followed the steps from https://www.youtube.com/watch?v=sUYVKhI_84E and use PiShrink, restors the card as `sudo dd if=sdcard_shrinked2.img of=/dev/mmcblk0 bs=4M; sync` and my Pi won't boot :(

Any clues appreciated ?

neilyoung, 2020-06-29:
If you have your SD plugged, what device is the SD card reader?

Check with

sudo lsblk --scsi

yuriw, 2020-06-29:
mmcblk0               179:0    0  29.7G  0 disk  
├─mmcblk0p1           179:1    0   256M  0 part  
└─mmcblk0p2           179:2    0   3.1G  0 part  /media/yuriw/rootfs

mmcblk0p1 shoud say "boot"

yuriw, 2020-06-29:
maybe balenaEtcher will work?

neilyoung, 2020-06-29:
Hmm. Not familiar with those kind of devices. Usually using /dev/sdX...

Balena Etcher works definitly.

This is how I restore usually:

// Just once
sudo apt-get install -y pv

// Unmount all sdXn

sudo mount | grep sdX
sudo umount /dev/sdX1
sudo umount /dev/sdX2
sudo umount /dev/sdX3

// Then
sudo pv -tpreb shrank-image.img | sudo dd of=/dev/sdX

yuriw, 2020-06-29:
Yup Balena Etcher works definitely !

Very odd the `dd` did not ....

yuriw, 2020-06-30:
wonder if pishrink can take *.gz file as input ?

Drewsif, 2020-07-01:
Glad the project works for you!

Interesting about the RO filesystem, I never really thought about that case. But I suppose this is why the `-s` options exists!

@yuriw There is no plan to support restoring since dd does that bit. Also it can not take `.gz` or any other compressed file as input. It would need to decompress it anyway to work on the image so there is no sense adding support.

closed 2020-07-01 by Drewsif