Drewsif/PiShrink #150: "parted: invalid token: primary Error: Expecting a file system type." 2 images

Conan179, 2020-07-03:
Hello@all

My Sd card from my pi are broken, with tet disk i can make 2 image.dd files from the /boot and the ext4 partion from the Card.
The main card was 64gb large, now I want to reduce the size of the 64gb image file.
How do I do that?
A simple ./pishrink.sh -vd /mnt/f/assas/pishrink.bin does nothing.

`pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
Line 296
beforesize: 60G
parted_output: BYT;
/mnt/f/assas/pishrink.bin:63591940096B:file:512:512:loop::;
1:0B:63591940095B:63591940096B:ext4::;
partnum: 1
partstart: 0
parttype: primary
tune2fs_output: tune2fs 1.45.5 (07-Jan-2020)
Filesystem volume name:   rootfs
Last mounted on:          /tmp/tmp.7atigx17sv
Filesystem UUID:          8f2a74a4-809c-471e-b4ad-a91bfd51d7c3
Filesystem magic number:  0xEF53
Filesystem revision #:    1 (dynamic)
Filesystem features:      has_journal ext_attr resize_inode dir_index filetype extent flex_bg sparse_super large_file dir_nlink extra_isize
Filesystem flags:         unsigned_directory_hash 
Default mount options:    user_xattr acl
Filesystem state:         clean
Errors behavior:          Continue
Filesystem OS type:       Linux
Inode count:              812800
Block count:              3244676
Reserved block count:     133252
Free blocks:              99790
Free inodes:              583647
First block:              0
Block size:               4096
Fragment size:            4096
Reserved GDT blocks:      420
Blocks per group:         32768
Fragments per group:      32768
Inodes per group:         8128
Inode blocks per group:   508
Flex block group size:    16
Filesystem created:       Thu Feb 13 17:30:14 2020
Last mount time:          Fri Jul  3 14:38:44 2020
Last write time:          Fri Jul  3 14:39:52 2020
Mount count:              0
Maximum mount count:      -1
Last checked:             Fri Jul  3 14:38:44 2020
Check interval:           0 (<none>)
Lifetime writes:          414 GB
Reserved blocks uid:      0 (user root)
Reserved blocks gid:      0 (group root)
First inode:              11
Inode size:	          256
Required extra isize:     32
Desired extra isize:      32
Journal inode:            8
Default directory hash:   half_md4
Directory Hash Seed:      8e66eea2-36ac-4fec-b5f4-bb012aa2cfca
Journal backup:           inode blocks
currentsize: 3244676
blocksize: 4096
pishrink.sh: Checking filesystem ...
rootfs: Inode 7272 extent tree (at level 1) could be narrower.  IGNORED.
rootfs: Inode 45097 extent tree (at level 2) could be narrower.  IGNORED.
rootfs: Inode 46904 extent tree (at level 2) could be narrower.  IGNORED.
rootfs: Inode 520487 extent tree (at level 2) could be narrower.  IGNORED.
rootfs: 229153/812800 files (0.4% non-contiguous), 3144886/3244676 blocks
resize2fs 1.45.5 (07-Jan-2020)
Line 325
currentsize: 3244676
minsize: 3239630
Line 333
extra_space: 5046
Line 340
minsize: 3244630
pishrink.sh: Shrinking filesystem ...
resize2fs 1.45.5 (07-Jan-2020)
Resizing the filesystem on /dev/loop2 to 3244630 (4k) blocks.
Begin pass 2 (max = 46)
Relocating blocks             ----------------------------------------XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 100)
Scanning inode table          ----------------------------------------XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop2 is now 3244630 (4k) blocks long.

Line 359
partnewsize: 13290004480
newpartend: 13290004480
parted: invalid token: primary
Error: Expecting a file system type.
pishrink.sh: ERROR occured in line 370: parted failed with rc 1
`

Can anybody help me, please?
Greetz Conan

framps, 2020-07-03:
Remove quotes in [this line](https://github.com/Drewsif/PiShrink/blob/master/pishrink.sh#L367) around `$parttype`

Conan179, 2020-07-03:
No canches, same error
[image]

framps, 2020-07-03:
Hm ... add following line `set -x` before the parted call and `set +x` after the parted call to see the actual  command executed.

Conan179, 2020-07-03:
[image]
The filesystem on /dev/loop3 is now 3240611 (4k) blocks long.

Line 359
partnewsize: 13273542656
newpartend: 13273542656
+ parted -s /mnt/f/assas/pishrink.bin unit B mkpart primary 0 13273542656
parted: invalid token: primary
Error: Expecting a file system type.
+ set +x
pishrink.sh: Shrinking image ...
Line 386
endresult: 0
Line 423
aftersize: 0
pishrink.sh: Shrunk /mnt/f/assas/pishrink.bin from 60G to 0 ...

now, the file are empty...

framps, 2020-07-03:
man parted gives
```
mkpart part-type [fs-type] start end
                     Make a part-type partition for filesystem fs-type (if specified), beginning at start and ending at end (by
                     default in megabytes).  part-type should be one of "primary", "logical", or "extended".
```
and the parted command shown complies to this syntax. You get various hits when you search for `parted: invalid token: primary`. But I don't get an idea what's the root cause for your error message :disappointed:

Conan179, 2020-07-03:
I think, the probem is.
My imae are not the full card but only the 2nd partition from the SD card. The script assumes, however, that the iamge has two patents, the / boot and / formatted with ext4, where all the data is.

Drewsif, 2020-07-03:
Correct, you do need a full image of the sd card not just a single partition. I'm going to leave this ticket open to try and detect this case and print a better error message

Bacri-FR, 2021-03-03:
Hello.
thanks for your great tool @Drewsif.

I'm facing the exact same problem.
Any chance you would add this feature to shrink only a IMG of a partition ?
thxs in advance.

jasvinder1079, 2024-03-13:
same error at line 376.

Drewsif, 2024-10-05:
> Hello. thanks for your great tool @Drewsif.
> 
> I'm facing the exact same problem. Any chance you would add this feature to shrink only a IMG of a partition ? thxs in advance.

Currently no plans to implement that feature, it would require some pretty heavy lifting I think for not a lot of gain. The use case for PiShrink is normally to back up the whole SD card so you can get a smaller image of a bootable Pi to flash back on. Sorry 😢

closed 2024-10-05 by Drewsif