Drewsif/PiShrink #151: Filesystem was not expanded when image booted the first time

peracchi, 2020-07-07:
@Drewsif **Thanks so much for this tool!**

I used Etcher to write _ubuntu-20.04-preinstalled-server-arm64+raspi.img.xz_ to a **Samsung 128 GB EVO Plus SD card**.

After customize the system as I needed, I created an image with the command:

`$ sudo dd if=/dev/sdb of=ninja_ubuntu-20.04.img bs=1M status=progress ; sync`

Then I used PiShrink on this image with the following command and output:

```
$ sudo pishrink.sh -vZap ninja_ubuntu-20.04.img ninja_ubuntu-20.04.pishrink.img

pishrink.sh v0.1.2
pishrink.sh: Copying ninja_ubuntu-20.04.img to ninja_ubuntu-20.04.pishrink.img... ...
pishrink.sh: Gathering data ...
pishrink.sh: Syspreping: Removing logs, apt archives, dhcp leases and ssh hostkeys ...
pishrink.sh: Checking filesystem ...
writable: Inode 23130 extent tree (at level 1) could be shorter.  IGNORED.
writable: Inode 37014 extent tree (at level 1) could be shorter.  IGNORED.
writable: 144542/7753088 files (0.1% non-contiguous), 1248217/31194875 blocks
resize2fs 1.45.6 (20-Mar-2020)
pishrink.sh: Shrinking filesystem ...
resize2fs 1.45.6 (20-Mar-2020)
Resizing the filesystem on /dev/loop0 to 1002440 (4k) blocks.
Begin pass 2 (max = 317884)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 952)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 24843)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 1002440 (4k) blocks long.

pishrink.sh: Shrinking image ...
pishrink.sh: Using xz on the shrunk image ...
ninja_ubuntu-20.04.pishrink.img (1/1)
  100 %     931,1 MiB / 4.172,8 MiB = 0,223    20 MiB/s       3:25             
pishrink.sh: Shrunk ninja_ubuntu-20.04.pishrink.img.xz from 120G to 932M ...

```
Later, I used Etcher to write _ninja_ubuntu-20.04.pishrink.img_ to **another** Samsung 128 GB EVO Plus SD card.

Using this copy on the same Raspberry Pi 4, after boot I noticed that _the filesystem was not expanded when image booted the first time_.

Below is the output of login and some commands to show that the filesystem was not expanded to use the 128 GB SD card.

```
Welcome to Ubuntu 20.04 LTS (GNU/Linux 5.4.0-1013-raspi aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

  System information as of Tue Jul  7 09:34:34 -03 2020

  System load:  0.08              Temperature:            45.3 C
  Usage of /:   77.2% of 3.70GB   Processes:              135
  Memory usage: 7%                Users logged in:        0
  Swap usage:   0%                IPv4 address for wlan0: 10.22.22.101

 * "If you've been waiting for the perfect Kubernetes dev solution for
   macOS, the wait is over. Learn how to install Microk8s on macOS."

   https://www.techrepublic.com/article/how-to-install-microk8s-on-macos/

0 updates can be installed immediately.
0 of these updates are security updates.

ubuntu@ninja:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            1.9G     0  1.9G   0% /dev
tmpfs           380M  3.9M  376M   2% /run
/dev/mmcblk0p2  3.8G  2.9G  690M  81% /
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           1.9G     0  1.9G   0% /sys/fs/cgroup
/dev/loop0       49M   49M     0 100% /snap/core18/1708
/dev/loop1       62M   62M     0 100% /snap/lxd/14808
/dev/loop2       49M   49M     0 100% /snap/core18/1753
/dev/loop3       64M   64M     0 100% /snap/lxd/16050
/dev/loop4       24M   24M     0 100% /snap/snapd/7267
/dev/loop5       26M   26M     0 100% /snap/snapd/8147
/dev/mmcblk0p1  253M   98M  155M  39% /boot/firmware
tmpfs           380M     0  380M   0% /run/user/1000
ubuntu@ninja:~$
ubuntu@ninja:~$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop0         7:0    0  48.4M  1 loop /snap/core18/1708
loop1         7:1    0  61.3M  1 loop /snap/lxd/14808
loop2         7:2    0  48.4M  1 loop /snap/core18/1753
loop3         7:3    0  63.6M  1 loop /snap/lxd/16050
loop4         7:4    0  23.5M  1 loop /snap/snapd/7267
loop5         7:5    0  25.9M  1 loop /snap/snapd/8147
mmcblk0     179:0    0 119.3G  0 disk
├─mmcblk0p1 179:1    0   256M  0 part /boot/firmware
└─mmcblk0p2 179:2    0   3.8G  0 part /

```

Great job, **from 120 GB file to a 932 MB file!** 🥇

madAle, 2020-07-27:
Same here

+1

ceisma, 2020-08-11:
Same here, and I don't have a read-only file system. I'm running the latest OSMC on a Rpi3b+.

jrtrevino, 2020-09-02:
Same issue here. PiShrink compressed a ~32GB .img to 4.15GB, and I flashed it onto a 32GB sd card for th raspberry pi 4 for boot. It works fine, though I have 25.60 GiB of unallocated space on the 32GB card.

belveder79, 2020-09-18:
there are many reasons why auto-expand might or might not work... here are two of them:

1) if your root is not a partition on /dev/mmcblk0 but a partition on /dev/mmcblk2 (like on the RockPro64 for example)
2) the rc.local is not run automatically, as it is replaced by another mechanism (like /var/lib/dietpi/postboot.d on DietPi)

BigMuscle85, 2020-11-03:
I just find out if /etc/rc.local file does not exist, pishrink does not create new script to expand FS on the first boot.

beowulff, 2021-02-03:
I'm having the same issue.
I looked in the compressed .img and /etc/rc.local was there (in fact, it has some scripts that I need to run installed). But, there was no decompression code listed, so pishrink must be silently failing for some reason. Is there some way to add some debugging comments?

jlagermann, 2021-03-20:
Same issue here.  In my case there is no rc.local in my original image.  It looks like pishrink.sh is not creating it.

wintimes, 2022-05-14:
same issue here. i use wsl2 to bash pishrink and it won't expand

robjamsinterit, 2023-05-09:
Same issue

Stormlink, 2023-08-29:
I had the same issue trying to install Ubuntu Server 22.04.3 LTS to a RPi4B.  I solved it by modifying pishrink.sh as follows:

Added before the semicolon of line 81:
 || ! [[ -f "$mountdir/etc/rc.local" ]]

Changed lines 121 & 132 to not invoke rc.local if it does not exist, so these lines now read:
rm -f /etc/rc.local; cp -fp /etc/rc.local.bak /etc/rc.local 2>/dev/null && /etc/rc.local

I'll attach my modified pishrink.sh
[PIShrink.zip](https://github.com/Drewsif/PiShrink/files/12460007/PIShrink.zip)

commit aa14313d (2023-12-01): Fixes for rc.local issues

- Display warning when no existing /etc/rc.local is found
- Only try to run old /etc/rc.local if it actually exists

Closes #151
Closes #267
closed 2023-12-01 by Drewsif