Drewsif/PiShrink #155: Does not seem to work on smb share

yuriw, 2020-08-19:
File location as `/run/user/1000/gvfs/smb-share:server=pi-nas.local,share=omvbackup`

> sudo pishrink.sh backup-omv-19-Aug-2020_10-54-34.ddfull.img pishrink.sh 
> pishrink.sh v0.1.2
> pishrink.sh: ERROR occured in line 216: backup-omv-19-Aug-2020_10-54-34.ddfull.img is not a file...

It'd be nice if it was able to work.  Pls consider.

framps, 2020-08-20:
Please provide the output of 
```
file backup-omv-19-Aug-2020_10-54-34.ddfull.img
```

yuriw, 2020-08-20:
@framps 

file backup-omv-19-Aug-2020_10-54-34.ddfull.img 
backup-omv-19-Aug-2020_10-54-34.ddfull.img: DOS/MBR boot sector; partition 1 : ID=0xc, start-CHS (0x40,0,1), end-CHS (0x3ff,3,32), startsector 8192, 524288 sectors; partition 2 : ID=0x83, start-CHS (0x3ff,3,32), end-CHS (0x3ff,3,32), startsector 532480, 12964041 sectors

framps, 2020-08-20:
What's the result of following command ?

```
[[ ! -f "backup-omv-19-Aug-2020_10-54-34.ddfull.img" ]] && echo "is no file" || echo "is file"
```

yuriw, 2020-08-20:
@framps nothing

yuriw, 2020-08-20:
BTW it does not work when file location is /run/user/1000/gvfs/smb-share:server=pi-nas.local,share=omvbackup

After I moved it to `~/tmp` it worked fine 
FYI

framps, 2020-08-20:
> After I moved it to `~/tmp` it worked fine

Great you tested this alternate way to use pishrink. pishrink uses the bash -f test to check if the img passed in the call is a file. For some reasons I don't know this test fails if the file is mounted via gfvs. Looks like this no pishrink issue but a gfvs issue. 

For sure I know pishrink works if the file is mounted via nfs or smb by not using gfvs. Maybe you find an alternative way to mount the image file?

yuriw, 2020-08-20:
@framps sure makes sense, will try `fstab`

framps, 2020-08-29:
@yuriw Do you have any update?

yuriw, 2020-08-29:
@framps no 
I thought we said it's a limitation if the file is mounted via `gfvs`

framps, 2020-08-29:
Ok. Then please close this issue :wink:

closed 2020-08-29 by yuriw