Drewsif/PiShrink #157: "/etc not found, autoexpand will not be enabled" error

ayhlai, 2020-08-27:
Thanks for the great tool.

I use a very standard raspbian build on Pi 4.  The script is in  /usr/local/bin/  and img I am trying to shrink is in /home/pi

When I run "sudo pishrink.sh myimage.img" , it gives an error " /etc not found, autoexpand will not be enabled ...".   What is the problem?

pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
**pishrink.sh: /etc not found, autoexpand will not be enabled ...**
pishrink.sh: Checking filesystem ...
volumio_data: 19400/5371520 files (0.1% non-contiguous), 371809/7181392 blocks
resize2fs 1.44.5 (15-Dec-2018)
pishrink.sh: Shrinking filesystem ...
resize2fs 1.44.5 (15-Dec-2018)
Resizing the filesystem on /dev/loop0 to 398880 (4k) blocks.

thanks

framps, 2020-08-27:
> /etc not found, autoexpand will not be enabled ...". What is the problem?

The message complaints about a missing /etc directory which is standard on Linux. What's the output of following command on your system?

```
ls -la /etc | wc -l
```

ayhlai, 2020-08-28:
Hi , thanks for reply. This folder does exist and here's the output. I have tried with various distribution including volumio and raspbian and still same problem.

volumio@volumio:~$ ls -la /etc | wc -l
193

framps, 2020-08-28:
That's what I expected :cry:  Could you please run [this modified code](https://github.com/framps/PiShrink/blob/master/pishrink.sh) and post the output?

ayhlai, 2020-08-30:
Thanks very much for the help. The output of the scripted seemed same , I have also paste the log output below.  I am not sure what is the problem since I tried multiple builds are it's still the same.

** output of the script **
(I stopped it after seeing the /etc not found)

img
pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
hello
pishrink.sh: /etc not found, autoexpand will not be enabled ...
pishrink.sh: Checking filesystem ...
volumio_data: 19400/5371520 files (0.1% non-contiguous), 371809/7181392 blocks
resize2fs 1.44.5 (15-Dec-2018)
pishrink.sh: Shrinking filesystem ...
resize2fs 1.44.5 (15-Dec-2018)
Resizing the filesystem on /dev/loop1 to 398880 (4k) blocks.
Begin pass 2 (max = 61652)

**Here the pishrink.log:**

/dev/mmcblk0p2 on / type ext4 (rw,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=823904k,nr_inodes=103729,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup2 on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=39,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=191200k,mode=700,uid=1000,gid=1000)
//192.168.1.6/shared on /mnt/shared type cifs (rw,relatime,vers=3.1.1,cache=strict,username=andrew,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.1.6,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=4194304,wsize=4194304,bsize=1048576,echo_interval=60,actimeo=1,user=andrew)
/dev/loop1 on /tmp/tmp.XHiA7gshDI type ext4 (rw,relatime,stripe=1024)

Drewsif, 2020-08-30:
It will still work even if it does not find /etc. It looks like your using volumeio which appears to have a data partition at the end of the disk. So we can shrink that partition but it won't auto expand. You can manually expand it and everything should be peachy

closed 2020-08-30 by Drewsif