Drewsif/PiShrink #164: ran it on an  imac, what's wrong and how to solve it?

qllbee, 2020-09-20:
Hello,
I ran this command on an  imac with osx10.14.6,I want to know what's wrong and how to solve it? 

fantas-iMac:desktop fanta$ sudo pishrink.sh old.img
/usr/local/bin/pishrink.sh: line 11: declare: -A: invalid option
declare: usage: declare [-afFirtx] [-p] [name[=value] ...]
/usr/local/bin/pishrink.sh: line 12: declare: -A: invalid option
declare: usage: declare [-afFirtx] [-p] [name[=value] ...]
/usr/local/bin/pishrink.sh: line 13: declare: -A: invalid option
declare: usage: declare [-afFirtx] [-p] [name[=value] ...]
pishrink.sh v0.1.2
pishrink.sh: ERROR occured in line 242: parted is not installed.
fantas-iMac:desktop fanta$

framps, 2020-09-20:
Either declare bash as your default shell or prefix the pishrink call with bash.

qllbee, 2020-09-20:
I think bash is the defalut shell

fantas-iMac:~ fanta$ ls -al `which sh`
-r-xr-xr-x  1 root  wheel  618480  7 17  2019 /bin/sh
fantas-iMac:~ fanta$ echo $SHELL
/bin/bash
fantas-iMac:~ fanta$

framps, 2020-09-20:
Well, than you have a bash version < 4. In v4 assoziative arrays were added (-A option of declare) .

qllbee, 2020-09-20:
I googled and updated the bash to 5.0.18
$ /bin/bash --version
5.0.18(1)-release (x86_64-apple-darwin18.7.0)

I ran it again,still have an error,parted is not installed?

fantas-iMac:desktop fanta$ sudo pishrink.sh big.img
pishrink.sh v0.1.2
pishrink.sh: ERROR occured in line 242: parted is not installed.
fantas-iMac:desktop fanta$

qllbee, 2020-09-20:
I don't know much about compiling.
I downloaded parted-3.3.
It doesn't seem to support MacOS,does it?

fantas-iMac:parted-3.3 fanta$ ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... build-aux/install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking build system type... x86_64-apple-darwin18.7.0
checking host system type... x86_64-apple-darwin18.7.0
configure: error: Unknown or unsupported OS "darwin18.7.0".  Only "linux", "uclinux", "gnu" and "beos" are supported in this version of GNU Parted.
fantas-iMac:parted-3.3 fanta$

framps, 2020-09-20:
> I don't know much about compiling.

I don't know much about macOS :cry:

Drewsif, 2020-09-25:
Parted is not supported on Mac, you will need to run pishrink in a VM or install Linux. There are some good youtube tutorials on getting pishrink installed inside a VM out there

closed 2020-09-25 by Drewsif

mgomesborges, 2020-09-25:
Hey @framps @Drewsif @qllbee you can use [PiShrink dockerized](https://github.com/mgomesborges/pishrink-docker) | [Docker Hub PiShrink](https://hub.docker.com/r/mgomesborges/pishrink):

1. [Install Docker Desktop on Mac](https://docs.docker.com/docker-for-mac/install/)

2. Make a copy of a Raspberry Pi SD card that you want to shrink ([see instructions here](https://github.com/mgomesborges/raspberry-pi/blob/master/setup/clone-sd-card.md)).

3. Using the Terminal, access the directory containing the Raspberry Pi image:

```bash
cd ~/Directory-with-RPi-image
```

4. Run PiShrink dockerized:

```bash
docker run --privileged=true --rm \
    --volume $(pwd):/workdir \
    mgomesborges/pishrink \
    pishrink -pZv IMAGE_NAME.img NEW-IMAGE_NAME.img
```

spidgorny, 2020-12-06:
This should be in README.md !!!

framps, 2020-12-06:
@spidgorny Why don't you just create a PR for the README?

spidgorny, 2020-12-07:
IMHO, it faster if the owner just copy/paste a piece of comment to README compared to the work required for third-party person that needs to:
* fork
* copy/paste
* check it looks nice
* create PR
* PR rejected because the author meant to have it in a different place in README
* repeat...

But I'll try...

mifi, 2023-08-18:
`pishrink -pZv` no longer works. it should be `pishrink -Zv`