Drewsif/PiShrink #168: Activate service to regenerate ssh host keys after said keys cleanup

gromain, 2020-10-17:
When PiShrink removes the hosts keys from the image, the said keys are not regenerated on the following first boot. The ssh service will then not be available, since there are no host keys.

One way to solve the problem would be to activate the `regenerate_ssh_host_keys` systemd service provided by Raspberry OS.

This could be done in the mounted filesystem by creating the following symlink:
`sudo ln -s /lib/systemd/system/regenerate_ssh_host_keys.service etc/systemd/system/multi-user.target.wants/regenerate_ssh_host_keys.service`.

I think #73 was linked to this issue.

alexanderfitu, 2021-07-04:
+1 on this. It renders a headless pi inaccessible. If you run the script without -p it does not accept the correct password.

Starwhooper, 2021-07-10:
> When PiShrink removes the hosts keys from the image, the said keys are not regenerated on the following first boot. The ssh service will then not be available, since there are no host keys.
>
> One way [...]

I can confirm. When i use pishrink with parameter -p, the ssh will not start when i use the image in further time.
To get the ssh service back, i use raspi-config to enable it again, but that's a fool solution ;-)

shatteredsword, 2022-10-10:
> When PiShrink removes the hosts keys from the image, the said keys are not regenerated on the following first boot. The ssh service will then not be available, since there are no host keys.
>
> One way [...]

Not sure what the status of this repo is, but I implemented your idea in a pull request. I tested it on the latest bullseye image and it works great!
[https://github.com/Drewsif/PiShrink/pull/244](https://github.com/Drewsif/PiShrink/pull/244)

shatteredsword, 2022-10-21:
currently working on a revision for non-raspbian OS images and possibly support for dropbear, cause why not

framps, 2022-10-21:
> Not sure what the status of this repo is,

Last commit is from 8/16/2021. As far as I can see it's no longer supported.

Drewsif, 2022-10-24:
Still alive, just focused on other things. Will eventually review your PR, I do appreciate the contribution though.

Drewsif, 2023-02-14:
I don't like the way cleanup is currently done and am looking into better options

closed 2023-02-14 by Drewsif