Drewsif/PiShrink #169: Some software does not want to see their log file disappear

gromain, 2020-10-17:
Cleanup the logs by removing everything inside `/var/log/` wreaks havoc for some programs that expect the files to be there.
Two are of note: Apache and Mosquitto. They both fail on startup if the log files or folders are not there.

Maybe a good way to solve this would be to only remove `.gz` logs and to empty existing files with `> /var/log/logfile.log` ?

gromain, 2020-10-17:
The suggested regex could even be changed to `.*log\.\([0-9]*\|gz\)$` to match any files whose names ends with either numbers or `gz` (to match on files like `error.log.1` for example)

gromain, 2020-10-17:
Also, the command `> /var/log/logfile.log` can be replace with `sudo find mnt/var/log/ -type f -regex '.*\.log$' -exec truncate -s0 {} +` which I think is more suited to a script.

LazeMSS, 2021-02-24:
Or you could use 
`find whateverfiles |xargs sudo truncate --size 0`

mtlynch, 2021-09-08:
nginx is another example of software that files if the log folder is removed:

```
Sep 03 16:30:56 tinypilot systemd[1]: Starting A high performance web server and a reverse proxy server...
Sep 03 16:30:56 tinypilot nginx[5229]: nginx: [alert] could not open error log file: open() "/var/log/nginx/error.log" failed (2: No such file or directory)
Sep 03 16:30:56 tinypilot nginx[5229]: 2021/09/03 16:30:56 [emerg] 5229#5229: open() "/var/log/nginx/error.log" failed (2: No such file or directory)
Sep 03 16:30:56 tinypilot nginx[5229]: nginx: configuration file /etc/nginx/nginx.conf test failed
Sep 03 16:30:56 tinypilot systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Sep 03 16:30:56 tinypilot systemd[1]: nginx.service: Failed with result 'exit-code'.
Sep 03 16:30:56 tinypilot systemd[1]: Failed to start A high performance web server and a reverse proxy server.
```

Drewsif, 2023-02-14:
I am looking into better ways to support cleanup. Closing this until thats done

closed 2023-02-14 by Drewsif