Drewsif/PiShrink #170: Feature Request: Copy Progress Bar

JamesM92, 2020-10-18:
I would like to have a progress bar for the copying function.

I don't like having it work on the file directly just encase something goes wrong, i don't want to have to create a new image so i prefer to tell pi-shrink to make a new file for it to work on. 

The issue with this is that it gives no indication of the progress it is making on the copying, if it is doing a 128gb image, you really have no clue how much longer its going to take before its ready.

framps, 2020-10-19:
pishrink uses the `cp` command to create a copy of the original file. cp doesn't support any progress indication. I found [this page](https://unix.stackexchange.com/questions/65077/is-it-possible-to-see-cp-speed-and-percent-copiedl) which discusses various ways to get cp to present some kind of progress information. As a mitigation you can use `progress` which was mentioned [here](https://unix.stackexchange.com/a/169211/95392) :wink:

elmot, 2020-10-26:
`dd status=progress` reports progress while copying

closed 2022-05-06 by JamesM92