Drewsif/PiShrink #183: PiShrink does not shrink file

Bboy486, 2021-01-10:
I have a windows share mounted on my Rpi4b.

Running 
wget https://raw.githubusercontent.com/Drewsif/PiShrink/master/pishrink.sh
chmod +x pishrink.sh
sudo mv pishrink.sh /usr/local/bin

sudo pishrink.sh /mnt/windowsnas/DenRaspberryPiBackup.img 

/mnt/windowsnas/DenRaspberryPiBackup-pishrink.img

I mounted the windows share in /mnt/windowsnas

However when running the command it doesn't shrink the file.

Drewsif, 2021-01-23:
Can you copy paste the full console output from trying to shrink it?

Bboy486, 2021-01-23:
> Can you copy paste the full console output from trying to shrink it?

What do I need to do to get that info to you?

Drewsif, 2021-02-10:
Run the command and copy and paste any output here

closed 2023-02-14 by Drewsif