Drewsif/PiShrink #184: pishrink doesn't work in crontab

meyerhaven, 2021-01-16:
Hey everybody,

I tried to back up my pi an shrink the image. Manually everything works fine and as expected, but if I put the script into crontab nothing happens. The part with dd works fine, but pishrink doesn't work. Everything i found was to pu a "-p" in the command, but this changed nothing.

Have you any ideas for me?

Here ist my script.

`#!/bin/bash
# VARIABLEN - HIER EDITIEREN
BACKUP_PFAD="/backup"
BACKUP_ANZAHL="8"
BACKUP_NAME="mailpiBackup"
# ENDE VARIABLEN

# Stoppe Dienste vor Backup
service fetchmail stop
service clamav-daemon stop
service clamav-freshclam stop
service amavis stop
service postfix stop
service cyrus-imapd stop

# Backup mit Hilfe von dd erstellen und im angegebenen Pfad speichern
sleep 5

dd if=/dev/mmcblk0 of=${BACKUP_PFAD}/${BACKUP_NAME}-$(date +%Y%m%d).img bs=1MB

sleep 5

# Starte Dienste nach Backup
service cyrus-imapd start
service clamav-daemon start
service clamav-freshclam start

service postfix start
service fetchmail start

rm /var/log/clamav/freshclam.log
freshclam

service amavis start

#Backup verkleinern
/usr/local/bin/pishrink.sh ${BACKUP_PFAD}/${BACKUP_NAME}-$(date +%Y%m%d).img

# Alte Sicherungen nach Anzahl neuer Sicherungen entfernen
pushd ${BACKUP_PFAD}; ls -tr ${BACKUP_PFAD}/${BACKUP_NAME}* | head -n -${BACKUP_ANZAHL} | xargs rm; popd
`

M80Falcon, 2021-01-27:
Hi,

same problem here. I created the cronjob with "sudo crontab -e". So it should work woth root permissions. The dd part is working well, but pishrink not. In a manual run with sudo everything works as expected. 

If you take a look in the log file of the cronjob, you will notice, that all of the required tools could not be found. But even if you skip the part of the check, for example parted will not be found, but it is installed and the path is also availabe.

Workaround:

Just edit the /etc/crontab file with "sudo nano /etc/crontab" and add the line "0  1    * * *   root    /usr/local/bin/Backup.sh" under the existing jobs. So the Backup.sh script will run every day at 1am. If you start the script from here, everything works fine.

meyerhaven, 2021-02-03:
Thank you, this worked for me, too :)
I don't understand the reason for this, but i,m happy this is working now!

closed 2021-02-03 by meyerhaven

The-Exterminator, 2021-03-29:
This work for me : 

https://github.com/The-Exterminator/PiShrink-to-Crontab