Drewsif/PiShrink #187: Creating sprase image before compressing

dilyanpalauzov, 2021-02-08:
In order to get smaller compressed image, place allocated by deleted files shall be zeroed.  This is called “sparse image”.  According to https://ext4.wiki.kernel.org/index.php/Ext4_VM_Images#Good_Tools_for_Making_Sparse_Images:_fstrim.2Fe2fsck.2Fzerofree to create a sparse image one has to call `e2fsck -E discard $loopback`.

• Call `e2fsck -E discard $loopback` on the resulting image.

jfikar, 2021-02-27:
The command `e2fsck -E discard $device` does not seem to discard the free space. Even `e2fsck -f -y -D -E discard $device` does not work.

On contrary, the `zerofree` of `dd if=/dev/zero of=zerofile;sync;sync;rm zerofile` works just fine.

There is also a possibility to create a sparse file at the end, if you are not compressing it by `fallocate -d image`

Drewsif, 2023-02-14:
Duplicate of #59

closed 2023-02-14 by Drewsif