Drewsif/PiShrink #188: PiShrink giving me an error on WSL Ubuntu.

TechNerd22, 2021-02-09:
I'm trying to shrink a Raspberry Pi OS 64-bit image, but when I attempt to shrink it, this happens

pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
tune2fs: No such file or directory while trying to open
Couldn't find valid filesystem superblock.
tune2fs 1.45.5 (07-Jan-2020)
pishrink.sh: ERROR occurred in line 289: tune2fs failed. Unable to shrink this type of image

How do I fix this? I'm running Ubuntu WSL on Win10 Pro 64-bit 20H2, the image is Raspberry Pi OS 64-bit img created using Win32DiskImager.

framps, 2021-02-09:
Looks like WSL is approx 50% compatible to Linux ... Just use a real Linux and you will be fine :wink:

TechNerd22, 2021-02-10:
> Looks like WSL is approx 50% compatible to Linux ... Just use a real Linux and you will be fine 😉

Would using an Ubuntu VM work fine? I'd rather do something like that instead of running Linux on an actual computer.

Drewsif, 2021-02-10:
A VM would work perfectly fine. Some people had issues using the Shared Folder feature in VirtualBox and VMware so if you run into issues make sure you're not using that.

closed 2021-02-10 by Drewsif