Drewsif/PiShrink #198: Unable to shrink dd image / error message

raviwarrier, 2021-05-31:
I get this error when I tried to shrink my img of the rpi sd card.

pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
Warning: Error fsyncing/closing /srv/dev-disk-by-uuid-E4FAC652FAC62122/RWPI_Backups/dd_images/2021-05-31.img: Input/output error
Error: Can't have a partition outside the disk!
pishrink.sh: ERROR occurred in line 273: parted failed with rc 1
pishrink.sh: Possibly invalid image. Run 'parted 2021-05-31.img unit B print' manually to investigate ...

I used dd to make an image of the sd card. Here's the command I used (if it helps):

`sudo dd bs=1M status=progress count=28900 if=/dev/mmcblk0 of=/srv/dev-disk-by-uuid-E4FAC652FAC62122/RWPI_Backups/dd_images/$filename`

am I doing something wrong? I just followed the instructions from https://www.tomshardware.com/how-to/back-up-raspberry-pi-as-disk-image

Cheers.

Drewsif, 2021-06-11:
What was the distro you had on it?

raviwarrier, 2021-06-11:
Raspbian

Jean-Reinhold, 2021-11-12:
I am having the issue! I guess that has something to do with dd, I copied and shrunk the same sd card before and it worked

geoffbrucepayne, 2022-01-27:
I had a similar issue.  Double check you are carefully following "Tom's" instructions and that your count is *larger* by at least several hundred MB than your resized partition - assuming you resized it using something like Gparted.  That was my issue I had the `count` too small on the `dd` command.  In my case I resized the partition to 4000 MiB, then used a count=5000 on `dd` and worked where it was failing before.

closed 2022-04-18 by Drewsif