Drewsif/PiShrink #199: Shrinked with PiShrink from 120 GB SSD, then written on SD card with 32GB and expanded to same size - give me an error when i write that 32 gb im back to ssd - why?

jayjupdhig, 2021-06-07:
-> https://files.slack.com/files-pri/T01JYNPBP2T-F023A1SKLH2/img_20210525_091756.jpg

Thx for feedbacks...

jayjupdhig, 2021-06-10:
No anwser?!?

jayjupdhig, 2021-06-10:
Ready for more??

HERE:

[image]
[image]
[image]
[image]
[image]
[image]

jayjupdhig, 2021-06-10:
-> When i write the image with "HDD Raw Copy Tool 1.10", i NEVER get got a problem [because of a problematik SSD or M2 drive]... so it is realistic that the issue was caused by your script!!

jayjupdhig, 2021-06-10:
For those who speak (or read) german:
https://forum-raspberrypi.de/forum/thread/52982-fsck-auf-raspbian-10-beim-booten-starten/?postID=494722#post494724

Drewsif, 2021-06-11:
1. I cant click your slack link, no idea what it even is
2. This is an open source project that I do in my free time, so I'm slow to respond
3. Calm down, being a twat won't get you support from me

jayjupdhig, 2021-06-11:
Hi. So you see, i've already closed this issue here. Thank you anyway...

jayjupdhig, 2021-06-11:
Additional info: Slack image is here. Just for correction:
[image]

jayjupdhig, 2021-07-21:
Problem still persists...

closed 2021-07-31 by Drewsif