Drewsif/PiShrink #200: Error Shrinking Image

brunomgrocha, 2021-06-25:
Hi,
I'm trying to run pishrink in a shell script that invokes pishrink. 

The script runs but stops when running pishrink with the following error:

pishrink.sh v0.1.2
pishrink.sh: ERROR occurred in line 242: parted is not installed.

Any ideia why this happens. Thank you

framps, 2021-06-25:
You have to install parted with `sudo apt-get install parted` :wink:

brunomgrocha, 2021-06-25:
Don't think that's the problem because if I run it outside of the initial script it works. The problem is when we run it inside the original script.

framps, 2021-06-25:
Hm ... that's interesting :open_mouth: . Try to replace https://github.com/Drewsif/PiShrink/blob/9d9c0dd32cd988f9c3bd9953ca245d33a52fc9e3/pishrink.sh#L240

with 
```
/usr/bin/which $command &>/dev/null
``` 

which is another way to check if a tool is available.

brunomgrocha, 2021-06-25:
still the same error
ERROR occurred in line 242: parted is not installed.

Running it directly works perfect,
pishrink.sh: Shrinking image ...
pishrink.sh: Shrunk /mnt/storage/backups_pi/2021-06-25.img from 30G to 4.8G ...

I try to install parted, and its installed. 
parted is already the newest version (3.2-25).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

To sum up,
Running it as a standalone it works,
under a script it outputs the error above

Running from cron reproduces the error above also.

framps, 2021-06-26:
That's strange. I call pishrink in a bash script and have no issue. Are you sure your code calls pishrink the right way?

brunomgrocha, 2021-06-26:
The code is this one
`
#!/bin/bash
. /home/bmmg/.bashrc

DestFolder="/mnt/storage/backups_pi/"
hname=`hostname`
BackupDir="${DestFolder}"

echo "Starting Backup"
dd if=/dev/sdb of=${BackupDir}/$(date +%Y-%m-%d).img bs=1M status=progress
echo "Backup Complete"
echo "Starting PiShrink"
source /home/xerxes/pibackup/pishrink.sh ${BackupDir}/$(date +%Y-%m-%d).img
`
Everything goes well until the last one "source....."

framps, 2021-06-26:
Is there any reason why you `source` pishrink? Did you try without `source`and just execute pishrink?

brunomgrocha, 2021-06-27:
No i didn't try that, but guess what....it works....
thank you for your support.
very appreciated.

framps, 2021-06-27:
You're welcome :wink: 

Frankly I don't have a clue what's the root cause it doesn't work with source but I'm glad it works now :smiley:

closed 2022-04-18 by Drewsif