Drewsif/PiShrink #201: PiShrink doesn't work when shrinking the Respberry Pi OS Lite released on May 7th 2021

qkpqkp, 2021-07-08:
PiShrink reports 'Error: image already shrunk to smallest size' in line 327.
PiShrink version: 0.1.2
resize2fs version: 1.45.5
GNU parted version: 3.3

lackdaz, 2021-07-08:
I'm also facing this issue. I've verified that the march release of Raspbian OS lite is working OK.

mfvargo, 2021-07-08:
I'm getting this issue.  Pulled the main branch of pishrink on an image I built today.

pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
pishrink.sh: Checking filesystem ...
rootfs: 60619/155040 files (0.2% non-contiguous), 444934/619335 blocks
resize2fs 1.45.6 (20-Mar-2020)
pishrink.sh: Shrinking filesystem ...
resize2fs 1.45.6 (20-Mar-2020)
Resizing the filesystem on /dev/loop48 to 616834 (4k) blocks.
The filesystem on /dev/loop48 is now 616834 (4k) blocks long.

parted: invalid token: primary
Error: Expecting a file system type.
pishrink.sh: ERROR occurred in line 370: parted failed with rc 1

closed 2023-02-07 by Drewsif