Drewsif/PiShrink #205: Odroid and other support?

yash-fn, 2021-08-03:
As a total idiot when it comes to filesystems, would this tool work for images created for other devices like the odroid series and etc? Like does it work for any image? Or are there certain assumptions about the filesystem used for the pi such that it only really works for the pi?

yash-fn, 2021-08-03:
So I tried it and it seems to work. The image compresses and when I flash it back onto another SD card, it runs. So I guess all is well? Just in case though, what are the assumptions for this script?

Drewsif, 2021-08-04:
It should work on any filesystem that resize2fs supports (ext2, ext3, ext4) as long as the last partition has all the free space. The automatic resize may or may not work depending on the OS but the shrinking should at least under those conditions.

closed 2021-08-04 by Drewsif