Drewsif/PiShrink #215: Support for non-sd-cards

vguna, 2021-10-04:
I'm backing up my raspberry system located on a SSD connected via USB to an .img file. Now I wanted to use pishrink on that image. But as far as I can see [here](https://github.com/Drewsif/PiShrink/blob/43f1a885cd147638a0d1961753f711f838915b88/pishrink.sh#L95) it seems only to support sdcards. Are there any plans to add support for other devices?

framps, 2021-10-06:
> Note: Both workarounds are untested.

I just tested #1. It doesn't work  :disappointed:

vguna, 2021-10-10:
Thanks, I will give #2 a try and report back!

RafigRzayev, 2022-10-12:
Hi, I also try to take an image of NVME SSD. I tried to follow the second approach. There were three occurrences of **_mmcblk0p_** and **_mmcblk0_** in the script, and I replaced them all with the word **_nvme_**, because when my system boots there is no sda, but nvme (nvme0n1p1, nvme0n1p2 and etc.)

After shrinking the image, when I flash it to a new system, following message appears during boot:

> rc.local[794]: nvme0n1p5 is not an SD card. Don't know how to expand. 

Up to now I have tried three options:
1. Using default script
2. Replacing mmcblk fields with sda
3. Replacing mmcblk fields with nvme
4. Currently testing replacing it with nvme0n1 but I doub't that it will have any positive impact. 

What should I do? Any help would be appreciated.

RafigRzayev, 2022-10-12:
@Drewsif any ideas? :)

framps, 2022-10-12:
Root cause is nvme uses a different naming scheme than mmcblk and the code [here](https://github.com/Drewsif/PiShrink/blob/43f1a885cd147638a0d1961753f711f838915b88/pishrink.sh#L91) fails and has to be adapted to the different naming scheme. (See also [here](https://unix.stackexchange.com/questions/452120/persistent-device-naming-for-nvme-storage-devices)). 

nmve has a namespace identifier prefixed by `n`:
`nvme0n1p5` which is nvme device 0, namespace 1 and partition 5. if this would be a sd card the name would be `mmcblk0p5`

RafigRzayev, 2022-10-12:
Thank you very much for the detailed information. I have two questions then:
1) Does it mean that my approach won't work directly? i.e. there is no word to replace mmcblk* in the script to make it work in my case. 
2) If so, could you please explain on what following line is trying to achieve?
`ROOT_PART=$(mount | sed -n 's|^/dev/\(.*\) on / .*|\1|p')`
If I understand what is expected output, maybe I can research and modify the script. 

For reference, below is structure of partitions on my system:

[image]

RafigRzayev, 2022-10-12:
This is result of latest attempt with nvme0n1:
[image]

framps, 2022-10-12:
raspi-config is a tool available on RaspberryOS. It's used to resize a partition. Until now I wasn't aware PiShrink is dependent on RaspberryOS :thinking:

RafigRzayev, 2022-10-13:
Ok, I was able to fix it and it works great! 

Need to change as follows:

```
  PART_NUM=${ROOT_PART#nvme0n1p}
  if [ "$PART_NUM" = "$ROOT_PART" ]; then
    echo "$ROOT_PART is not an SD card. Don't know how to expand"
    return 0
  fi
  # Get the starting offset of the root partition
  PART_START=$(parted /dev/nvme0n1 -ms unit s p | grep "^${PART_NUM}" | cut -f 2 -d: | sed 's/[^0-9]//g')
  [ "$PART_START" ] || return 1
  # Return value will likely be error for fdisk as it fails to reload the
  # partition table because the root fs is mounted
  fdisk /dev/nvme0n1 <<EOF
```
In other words, need to replace:

- **mmcblk0p** with **nvme0n1p**
- **mmcblk0** with **nvme0n1**

framps, 2022-10-13:
Thank you very much for sharing your solution. 

It's actually a hard coded workaround for your issue and should be generalized by @Drewsif . Not sure whether this will ever happen. Last commit is more than one year ago :disappointed: