Drewsif/PiShrink #220: Burned images not booting

sborn, 2021-11-29:
I have images that I've put on a flash drive and then moved over to a Ubuntu machine that I then shrink them on (no errors). When I go to boot from cards burned from those images, I get nothing. 

The images are of chilipie-kiosk, a version running on 3B+ that I've recently upgraded to Buster from Stretch.
Ubuntu is running on its own machine at 20.04.3 LTS

I'm burning on Win10 with Win32DiskImager

Images work that are burned using the original images.

sborn, 2021-11-29:
The boot partition files look correct from Win10, for what it is worth.

sborn, 2021-11-29:
I'm going to say that it might be the Ubuntu machine. I just shrunk a card using gparted and now that card won't work. I'll come back after troubleshooting on another machine tomorrow.

sborn, 2021-11-30:
Yeah, it wasn't the machine. A working build that I've used for over a year does the same thing. It happens with pishrink but also with gparted; if I shrink the image file I get a black screen, but if i get a working card and then shrink the partition with gparted it also gives me nothing.

moi427, 2021-12-08:
Same problem here and it isn't the machine. The shrinked image (from 32GB to 11GB) is apparently (green light) a few times read, but then nothing else happens. I even let it be in place for about 30 minutes, hoping that it would be because much time was needed to expand the image again. Bit no effect, nothing happens. A pitty, since it's ideal to move a lot smaller image around!

moi427, 2022-03-14:
I've been trying again, but unfortunately the problem remains the same: PiShrink works great in reducing the image size considerably, but when written to an SD card it's read a few times (green light) but that's it. So no working OS. I''ll put the output of the job below, maybe that helps since I'm out of clues and haven't been able to find something on the web to solve this (I'm not a Linux guy).

root
Ma 14-3-2022 08:33
pishrink.sh: Creating log file /root/pishrink.log ...
pishrink.sh v0.1.2
pishrink.sh: Copying /backup/Image-backup/RPI3-res.img to /backup/Image-backup/newpi.img... ...
pishrink.sh: Gathering data ...
Creating new /etc/rc.local
pishrink.sh: Checking filesystem ...
/dev/loop0: Inode 654089 extent tree (at level 1) could be shorter.  IGNORED.
/dev/loop0: 232307/1945888 files (0.3% non-contiguous), 2627075/7779968 blocks
resize2fs 1.44.5 (15-Dec-2018)
pishrink.sh: Shrinking filesystem ...
resize2fs 1.44.5 (15-Dec-2018)
Resizing the filesystem on /dev/loop0 to 2656006 (4k) blocks.
Begin pass 2 (max = 1590243)
Relocating blocks             ----------------------------------------XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 238)
Scanning inode table          ----------------------------------------XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 23199)
Updating inode references     ----------------------------------------XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 2656006 (4k) blocks long.

pishrink.sh: Shrinking image ...
pishrink.sh: Using pigz on the shrunk image ...
pishrink.sh: Shrunk /backup/Image-backup/newpi.img.gz from 30G to 4.1G ...

sborn, 2022-03-18:
> Same problem here and it isn't the machine. The shrinked image (from 32GB to 11GB) is apparently (green light) a few times read, but then nothing else happens. 

What OS/build are you writing an image of?

moi427, 2022-03-18:
I'm running Raspbian GNU/Linux 10 (buster)

DaveWhoHasARadarGun, 2022-03-21:
@sborn @moi427 Same issue, did either one of you find a solution?

DaveWhoHasARadarGun, 2022-03-21:
Thanks for your response - I just managed to get it to work, but I'm not sure if it's anything I did in particular or just a fluke. Here's my process, if it's of any use to anyone:

The first time that I used PiShrink, I did it on my Linux (Ubuntu20.04) laptop, and then used Raspberry Pi Imager to burn the shrunken .img file on that same laptop. The process to shrink the file was extremely fast, but the SD Card (16 GB) wouldn't boot. When I zipped the shrunken .img file, it went from 4.3 GB to 531MB. 

The second time I tried, I used WSL2 (Windows Subsystem Linux ver 2, with Ubuntu 20.04) on my Windows machine, and used the Raspberry Pi Imager (on my Windows machine this time) to burn the shrunken .img file to a different 16 GB SD Card. The process to shrink the file was much slower this time, and the resulting .img file was the same size (4.3 GB). When I zipped the file this time though, it only went down to 1.7 GB.  The SD Card booted without issue and all expected system functionality is working.

moi427, 2022-03-22:
That's curious anyway, I might try to burn a shrunken image (created on the RPi) that way.
So if I interpret your text correct, you ran PiShrink as usual on the RPi (so created the shrunken image on the RPi), but the burning that succeeded was by WSL2 on the Windows machine. Or did you also ran PiShrink under WSL2, shrinking the original image created on the RPi?

moi427, 2022-03-22:
Ok, thank you for confirming! What ran where wasn't entirely clear to me.
I already have WSL2 installed, but running PiShrink on my Windows machines wouldn't help me: then I first still would need to transfer the "normal" image of 30Gb to that Windows machine and that is just what I'm trying to avoid since transfer of this RPi3 isn't fast at all. Very slow in fact...

Drewsif, 2023-02-14:
Im pretty sure this is related to #255, the symptoms sound the same. So I'm going to close this as a duplicate until we confirm its not that

closed 2023-02-14 by Drewsif

moi427, 2023-02-14:
Well, unfortunately I never found a working solution using PiShrink.
If it is like in the link in #255 that the partition UUID (PARTUUID) is changed, that then might be the cause but then still PiShrink still isn’t working is it? I never got it working anyway, that is it works perfect to make a small image, but that can’t be booted from. So the real usage then isn’t there. That’s a real pitty.

Drewsif, 2023-02-14:
The problem is in usage of Parted not just PiShrink. The reported issue on stackexchange was related to using Parted and GParted, not even via PiShrink. Lots of people are still having success using PiShrink and Parted so I think it has to be version related.

You could try running on a different operating system. But if you wanted to go post in #255 what version of operating system you are using and what version of Parted you have that would be helpful in hunting hit down.

BillYovino, 2024-07-15:
I'm having the same issue trying to copy a Bullseye image from a 32gb SD card to a 16gb USB drive.

Pi Zero 2 W
Windows 11
WSL Ubuntu
PiShrink
Balena Etcher
Raspberry Pi Imager
Samsung 32gb Evo microSD
Sandisk 16gb USB drive

The shrunken image looks fine in Windows.

I've tried burning the image using Balena Etcher and Raspberry Pi Imager. For the most part nothing happens, but one time (using RPi Imager) it booted to a login prompt but wouldn't take my password. It seems like Imager used a 24 or more character string for the password even though I had it set up the way I wanted. No other attempts before or since have gotten to that point.

After 5 days of frustration I just ordered larger USB drives, but I would still love to know how to get this to work.

moi427, 2024-07-16:
I never got it running, just like my last comment in 2023. Couldn't find a relation to mentioned problem with Parted or GParted like in mentioned link  [https://github.com/Drewsif/PiShrink/issues/255](url) and from there to [https://raspberrypi.stackexchange.com/questions/68082/why-wont-my-raspberrypi-boot-if-i-use-parted-to-adjust-the-partition](url). So I stopped trying, 
I'm not a Linux guy, don't have enough knowledge of it so just try to have simple automated jobs running without needed manual intervention.
Definitely a pitty!