Drewsif/PiShrink #223: PiShrink does not ummount or release mounted image after shrink?

Jibun-no-Kage, 2021-12-24:
PiShrink does not ummount or release mounted image after shrink?

# /shrink# mount | grep rootfs
/shrink/2021-12-22 Pi3ModelB0 (Real Time Clock Source Example).img on /media/root/rootfs type ext4 (rw,nosuid,nodev,relatime,errors=remount-ro,uhelper=udisks2)

Should PiShrink not release the mounted image?

Drewsif, 2022-04-18:
It should unmount the image once it is done with the shrink. Did PiShrink finish successfully?

Jibun-no-Kage, 2022-04-19:
Right, it is when there is an error or unsuccessful result, the mount point remains.  So whatever clean up should be working, seems to fail at least some of the time.  Since efsck tends to choke on non linux partition, i.e. boot being vfat, that is on such trigger for the abort and no clean up, based on my experience thus far.

Drewsif, 2022-04-20:
Ahh that makes sense. Ill leave this open to remember to look into making sure we clean up better

Jibun-no-Kage, 2022-04-20:
Cool.  That is what I was looking for.  Thanks.

Drewsif, 2024-10-05:
Checked it again and it should be good

closed 2024-10-05 by Drewsif