Drewsif/PiShrink #225: Error occurred in line 216:  pi.img is not a file

mtugrulesin, 2022-01-29:
```
pi@raspberrypi:~ $ sudo pishrink.sh /home/pi/pi.img
pishrink.sh v0.1.2
pishrink.sh: ERROR occurred in line 216: pi.img is not a file
```

how can I fix this issue ?

arm358, 2022-02-14:
I was having issues with the way I was making my .img file. I was using MacOS to do it in DiskUtility and no matter what PiShrink didn't like the image that was produced (even though it worked fine to flash onto an SD card).

This should work if you're running Linux though (the only thing that ended up working for me):
sudo dd bs=4M if=/dev/{SDX} of=/home/{USER}/{FILENAME}.img status=progress

replace {SDX} with the mount point of your SD Card containing RPiOS (mine was /dev/sde, find this by using lsblk command)
replace {USER} with your username
replace {FILENAME} with whatever filename you want.

Then once that imager finishes you should be able to use PiShrink

Drewsif, 2022-04-18:
Lots of things can cause this issue. It's normally an artifact of how you are running PiShrink, if you are using a shared folder in VirtualBox that is your issue

closed 2022-04-18 by Drewsif