Drewsif/PiShrink #230: Pi OS 11 Bullseye OpenSSH server broken by PiShrink!  Delete of host keys Breaks OpenSSH Server?

Jibun-no-Kage, 2022-02-11:
Pi OS 11 Bullseye OpenSSH server broken by PiShrink!  Delete of host keys Breaks OpenSSH Server?  Used pishrink to remove files option, and once flash image and boot, OpenSSH refuses to run, fails completely.  Error reads as 'No hostkeys available -- exiting.'

framps, 2022-02-11:
Looks like you use an old pishrink version. Current version deletes ssh keys only if option `-p` is used.

Jibun-no-Kage, 2022-02-11:
I am using the latest.  My point is, there is no warning of how it breaks SSH connectivity.  Suggesting you add a comment to the help screen, like 'This option breaks SSH communication'

framps, 2022-02-11:
I see your point. But https://github.com/Drewsif/PiShrink/blob/43f1a885cd147638a0d1961753f711f838915b88/pishrink.sh#L166 writes for option `-p`: 

`Remove logs, apt archives, dhcp leases and ssh hostkeys`

No ssh hostkeys - no ssh connectivity :smirk:

Jibun-no-Kage, 2022-02-11:
Yeah... just thinking it would be user friendly.

closed 2022-02-11 by Jibun-no-Kage

gnulux, 2022-04-05:
I'm ok with -p option and it description but if i really need to delete ssh_host* keys and recreate then during the first boot for each new host i'm flashing. 
It could be good to control if the files exist overwhise recreate them.
I have many hosts  to flash and sshd is enabled by default in my image. Don't want to have the samer host key on all hosts.

Drewsif, 2022-04-20:
Some distros automatically regen the sshd keys if they are missing, which is why it just worked in testing. I would recommend setting up a custom cleaning script before you take images if you want that feature and the `-p` option does not work. You can also set your image to regen the keys on its boot if they are missing

I have been thinking about ways to make custom cleaning options but its still in the works

Jibun-no-Kage, 2022-04-20:
Yeah, that is cool.