Drewsif/PiShrink #234: Support for PINN

myxxmikeyxx, 2022-05-06:
This is more of a feature request but could you support PINN OS.
Its like NOOBS OS but have more features. The problem is that the headless setup doesn't want to connect to Wi-Fi so I am manually connecting it before installing any images. the problem is that when I try to run pishrink it says
`pishrink.sh: Shrunk pishrank_pinn.img from 30G to 30G ... `

Total output
`sudo bash ./pishrink.sh -vrap custom_pinn_backup_with_wifi.img pishrank_pinn.img            pishrink.sh v0.1.2                                                                     pishrink.sh: Copying custom_pinn_backup_with_wifi.img to pishrank_pinn.img... ...      pishrink.sh: Gathering data ...                                                        WARNING: PiShrink does not yet support autoexpanding of this type of image             pishrink.sh: Syspreping: Removing logs, apt archives, dhcp leases and ssh hostkeys ... pishrink.sh: Checking filesystem ...                                                   SETTINGS: 105/8192 files (0.0% non-contiguous), 2916/32764 blocks                      resize2fs 1.45.5 (07-Jan-2020)                                                         pishrink.sh: Shrinking filesystem ...                                                  resize2fs 1.45.5 (07-Jan-2020)                                                         Resizing the filesystem on /dev/loop0 to 7976 (1k) blocks.                             Begin pass 2 (max = 1460)                                                              Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX                 Begin pass 3 (max = 4)                                                                 Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX                 Begin pass 4 (max = 35)                                                                Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX                 The filesystem on /dev/loop0 is now 7976 (1k) blocks long.                                                                                                                    pishrink.sh: Shrinking image ...                                                       pishrink.sh: Shrunk pishrank_pinn.img from 30G to 30G ... `

Drewsif, 2023-02-07:
If I'm reading the output right, I can not easily add support for that type of partition layout. If the last partition is not the one with all the free space the script won't do much and at the moment I'm not positive how to support them.

closed 2023-02-07 by Drewsif

myxxmikeyxx, 2023-02-07:
Not a problem. I have since started using multiple small SD cards. Sorry for any trouble.

Drewsif, 2023-02-14:
No worries 😄