Drewsif/PiShrink #235: gzip failed

thealex626, 2022-05-18:
Hi! I've been shrinking a 64 GB image with -z argument. Here's the output:

`$ sudo pishrink.sh -z osmc.img osmc-shrinked.img
pishrink.sh v0.1.2
pishrink.sh: Copying osmc.img to osmc-shrinked.img... ...
pishrink.sh: Gathering data ...
Creating new /etc/rc.local
pishrink.sh: Checking filesystem ...
/dev/loop8: 51693/3883008 files (0.1% non-contiguous), 736718/15510016 blocks
resize2fs 1.45.5 (07-Jan-2020)
pishrink.sh: Shrinking filesystem ...
resize2fs 1.45.5 (07-Jan-2020)
Resizing the filesystem on /dev/loop8 to 798465 (4k) blocks.
Begin pass 2 (max = 476229)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 474)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 11304)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop8 is now 798465 (4k) blocks long.

pishrink.sh: Shrinking image ...
pishrink.sh: Using gzip on the shrunk image ...
gzip: osmc-shrinked.img.gz: Value too large for defined data type
pishrink.sh: ERROR occurred in line 421: gzip failed with rc 0
`

The archive is still created, although I don't know if the image is fine.

Drewsif, 2023-02-07:
Without the `-z` option is the image usable?

Drewsif, 2023-02-15:
Ok! Not really sure why gzip would error, the script just calls gzip directly so I have no theory as to why it would have broken.

closed 2023-02-15 by Drewsif