Drewsif/PiShrink #238: PiShrink not working with Home Assistant

EtienneStieger, 2022-06-20:
Hi,

PiShrink does not work properly with Home Assistant (Hassio).
It does the shrink without complaining (via Windows Subsystem for Linux - 5.10.16.3-microsoft-standard-WSL2) , but the momemt I write the shrunk image to SD card on Win 10 using Balena Etcher (or Win32DiskImager), the resulting SD card does not work.

A interesting side-effect to this is that the SD card maps to a handfull of drive letters on Windows 10, instead of the usual 2 drive letters.

Can this issue perhaps be looked into? It would be greatly appreciated.

GeoffMorg, 2022-06-26:
I have the same problem. In fact I thought it had trashed the SD card as it wouldn't even mount back on the PC I had written it on - but it did on another, and you could see all these partitions, and it mounted correctly. No working boot partition though. This is when writing it on a WSL 2 UBUNTU, which completes with no errors.

I have problems writing on another Pi too - telling me it won't autoexpand due to missing /etc - but that's another issue. Shame though as pishrink was invaluable for helping me use a different SD card on a number of occassions.

Happy to provide any necessary assistance to debug this.

Drewsif, 2023-02-07:
Can you post the output of the script when you run it?

closed 2024-10-05 by Drewsif