Drewsif/PiShrink #239: pishrink.sh: ERROR occurred in line 216: pi.img is not a file...

nituladaboneko, 2022-06-26:
pi@raspberrypi:/usr/local/bin $ sudo pishrink.sh pi.img
pishrink.sh v0.1.2
pishrink.sh: ERROR occurred in line 216: pi.img is not a file...

jfitzpat, 2022-07-17:
I got exactly the same error.

framps, 2022-07-17:
What's the result of 
```
la -la pi.img?
```

framps, 2022-07-18:
Hm ... according [this description](https://stackoverflow.com/a/10204597/1742425) 

```
-f file
              True if file exists and is a regular file.

A regular file is something that isn't a directory, symlink, socket, device, etc.
```

you shouldn't get the error message :confused:

jfitzpat, 2022-07-18:
It is very peculiar. The image in on a mounted thumb drive. I tried gzip, then reloaded it on another SD card with the PI Imager - nothing weird about the image. I've also used successfully used pishrink.sh before...

shatteredsword, 2022-10-10:
pishrink doesn't work correctly if run directly on an external drive. you'll need to copy the file over to your filesystem and then back again once you've shrank the file.

BigBobDalton, 2023-01-24:
I get the same error.  No, its not on a external drive.

Drewsif, 2023-02-07:
Try giving the full path to the file instead of a relative path. If there are spaces in the path be sure to put it in quotes

closed 2023-02-07 by Drewsif