Drewsif/PiShrink #243: Bad magic number in super-block while trying to open loop device

JohanNicander, 2022-09-30:
### Setup
**OS**: Ubuntu 20.04.5 LTS (bare metal)
**sd-card OS**: Raspberry Pi OS Light (64-bit)

### Steps to reproduce error
Clone the sd card (same results using `dd`)
```shell
$ pv < /dev/sda > my-image.img
```
Run PiShrink
```shell
$ pishrink.sh my-image.img
pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
tune2fs: Bad magic number in super-block while trying to open /dev/loop6
tune2fs 1.45.5 (07-Jan-2020)
pishrink.sh: ERROR occurred in line 298: tune2fs failed. Unable to shrink this type of image
```

### Extra info
```shell
$ file my-image.img
my-image.img: DOS/MBR boot sector; partition 1 : ID=0xc, start-CHS (0x40,0,1), end-CHS (0x3ff,3,32), startsector 8192, 524288 sectors; partition 2 : ID=0x83, start-CHS (0x3ff,3,32), end-CHS (0x3ff,63,32), startsector 532480, 30384128 sectors
```
```shell
$ fdisk -l my-image.img
Disk my-image.img 14.74 GiB, 15829303296 bytes, 30916608 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x8b97fb85

Device         Boot  Start  End      Sectors  Size  Id Type
my-image.img1        8192   532479   524288   256M  c  W95 FAT32 (LBA)
my-image.img2        532480 30916607 30384128 14.5G 83 Linux
```

```shell
mount -t ext4 my-image.img /mnt -o loop,offset=272629760,ro
mount: /mnt: wrong fs type, bad option, bad superblock on /dev/loop2, missing codepage or helper program, or other error.
```

fredrikfalkman, 2022-10-13:
I have had the same problem running Pishrink on Mac, Ubuntu and Raspian. Also trying to compress Raspberry Pi OS Light (64-bit) 24Gb SD card. 

[image]

Drewsif, 2023-02-14:
So that error usually means the OS does not know what the partitions filesystem is. Can anyone confirm if the partition layout changed in the latest Raspberry Pi OS Light?

develcore, 2023-04-02:
> Can anyone confirm if the partition layout changed in the latest Raspberry Pi OS Light?

Ich benutze Raspbian Buster. Aber der selber Fehler.

```
mmcblk0     179:0    0  7.3G  0 disk
|-mmcblk0p1 179:1    0  128M  0 part /boot
`-mmcblk0p2 179:2    0  7.2G  0 part /
```

mmcblk0p1 ist fat32.
mmcblk0p2 ist f2fs, nicht ext4.

firstchanceongit, 2024-12-07:
Getting the same error on most current version of Raspberry OS 64 lite (64-bit) (Bookworm)

1. Win32DiskImager used to create image from SD-Card with fresh installation of Raspberry OS with Raspberry Pi Imager)
2. pishrink installed as per Linux Installation instructions on https://github.com/Drewsif/PiShrink
3. Run command: sudo pishrink.sh /mnt/hdd/backup/file.img
4. Output:

> PiShrink v24.10.23 - https://github.com/Drewsif/PiShrink
> 
> pishrink.sh: Gathering data
> tune2fs: Bad magic number in super-block while trying to open /dev/loop0
> tune2fs 1.47.0 (5-Feb-2023)
> pishrink.sh: ERROR occurred in line 319: tune2fs failed. Unable to shrink this type of image

Can you please look into this? I'm at a loss here. pishrink used to be extremely helpful in the past - and will hopefully continue be in the future ...?

MikeGranby, 2025-01-22:
I'm seeing the same thing, and it seems to be that parted isn't listing the second partition...