Drewsif/PiShrink #245: Short read while trying to resize - automatically run e2fsck?

mrsilver76, 2022-10-26:
When I try to resize my image I get the following:

```
pishrink.sh: Checking filesystem ...
rootfs: 227270/1867728 files (0.2% non-contiguous), 2563156/7634688 blocks
resize2fs 1.46.2 (28-Feb-2021)
pishrink.sh: Shrinking filesystem ...
resize2fs 1.46.2 (28-Feb-2021)
Resizing the filesystem on /dev/loop1 to 2597468 (4k) blocks.
Begin pass 2 (max = 1050994)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXresize2fs: Attempt to read block from filesystem resulted in short read while trying to resize /dev/loop1
Please run 'e2fsck -fy /dev/loop1' to fix the filesystem
after the aborted resize operation.
pishrink.sh: ERROR occurred in line 355: resize2fs failed with rc 1
mount: /tmp/tmp.0eFiyUIJc6: can't read superblock on /dev/loop1.
mv: cannot stat '/tmp/tmp.0eFiyUIJc6/etc/rc.local.bak': No such file or directory
umount: /tmp/tmp.0eFiyUIJc6: not mounted.
pi@home:~ $ 
```

Would it be possible to get PiShrink to automatically run `e2fsck -fy /dev/loop1` should `resize2fs` make that recommendation?

Drewsif, 2023-02-14:
It should have checked the filesystem before running the command. If you run the command manually does it allow the image to shrink?

mrsilver76, 2023-02-25:
I did try, but e2fsck reported something about the superblock not being readable.

Sorry I can't be much more help, it was quite a while ago that I tried.

Drewsif, 2023-03-04:
It was my bad for the late reply. My assumption is the image was too messed up for it to be fixed. But should you run into it again feel free to open another issue.

closed 2023-03-04 by Drewsif