Drewsif/PiShrink #246: "Headless Re-Configure WiFi" after PiShrunk For Ubuntu Server 22.04 on Raspberry Pi

slowrunner, 2022-10-27:
I have created a Raspberry Pi image - "ROS2 Humble on Ubuntu 22.04 Server Jammy For GoPiGo3 Robots", and used "sudo PiShrink -p -v -z 32GB.img Distro2G.img" to shrink the image for distribution.

Using Raspberry Pi Imager v1.7.3 to write my image (with the setup option to set WiFi SSID/Password, userid/pswd, enable ssh, hostname, and local) creates a working system, but while cloud-init does copy the new SSID/passwd to /etc/netplan/50-cloud-init.yaml the content of the file is not used to configure the WiFi.

The system connects to the network SSID/password of the original image somehow.

I tried sudo cloud-init clean --logs before I shutdown the original image for shrinking, but that did not cause the new network-config content to be used on the redistributed image.

Any tips on making a redistributable headless configurable Ubuntu 22.04 Server image?

Drewsif, 2023-02-07:
This sounds like an issue with Raspberry Pi Imager, not PIShrink as we don't mess with anything related to wifi. Unless they also use `/etc/rc.local` they should work together.

closed 2023-02-07 by Drewsif

slowrunner, 2023-02-07:
What I found to work is performing the following before the shrink:
- sudo cloud-init clean --logs
- rm /etc/netplan/50-cloud-init.yaml