Drewsif/PiShrink #248: feature request: allow to shrink directly on the sd card

ilyapashuk, 2022-11-12:
hello

feature request:
please make it possible to shrink image directly on the sd card.

steps:
1. shrink the filesystem and shrink the partition on the sd card directly (sd card is attached to the pc with the card reader).
2. copy the shrunked image to the pc.
3. optional: revert the changes on the sd card.
4. do other stuff, like ssh keys cleanup or autoexpend setup, on the copied image.
5. compression

if the large sd card (32 gb or more) is used with the small data size (2-3 gb), disk space (and possibly speed) benefit may be dramatic.

framps, 2022-11-12:
> if the large sd card (32 gb or more) is used with the small data size (2-3 gb), disk space (and possibly speed) benefit may be dramatic.

Why don't you just allocate a partition of 2-3GB - or about 8GB?

fedekrum, 2022-11-18:
> Why don't you just allocate a partition of 2-3GB - or about 8GB?

Sometimes you do images from other installations that by default will expand the partition to the maximum size. So you finish having a full SD size.

I up vote @ilyapashuk request.
It will be a nice feature !!

shatteredsword, 2023-02-04:
this seems way outside the scope of this script.

Drewsif, 2023-02-07:
Im going to agree that it's outside of the scope of the project. I don't want to have the script making potentially breaking changes directly to the SD card. Working off a backup is much safer, if I can think up a way to do it safely I will reconsider but for now I don't think it will be added

closed 2023-02-07 by Drewsif