Drewsif/PiShrink #269: Raspbian not completly booting - No GUI underscore blinking

davidzfr, 2023-11-23:
Hi, I've the same problem as issue 73.

pishrink script was used without any option. No error message. Size was clearly reduced.
shrinked image was put back on a another SD card with win32diskimager.

Raspbian 12 desktop 32 bits.
No GUI, no access to desktop on the screen.
Only an underscore blinking at the top left corner of the screen.
I can access to the Rpi through ssh.
The config.txt is exactly the same.

Any Idea ?
Regards, David

_Originally posted by @davidzfr in https://github.com/Drewsif/PiShrink/issues/73#issuecomment-1823929684_

davidzfr, 2023-11-24:
With the same SD card...
I've tried with a Raspbian 11 - 64 bits -> OK
I've tried with an another script (shrinkwrap) on Raspbian 12 - 32 bits -> same problem as PiShrink

Drewsif, 2024-10-05:
My best guess is it's this issue https://github.com/Drewsif/PiShrink/issues/255. I don't see anything weird in the output. Sorry, hopefully I can get something to address that soon

closed 2024-10-05 by Drewsif