Drewsif/PiShrink #276: ERROR occurred in line 304: tune2fs failed

ROQUI101, 2024-05-17:
Dear sirs,

I have been trying to compress several images and I always run into this error:

tune2fs: Bad magic number in super-block while trying to open /dev/loop0
tune2fs 1.46.5 (30-Dec-2021)
/dev/loop0 contains a exfat file system
pishrink.sh: ERROR occurred in line 304: tune2fs failed. Unable to shrink this type of image

Any idea about what is the problem?

[image]

Thank you

borgesnotes, 2024-05-17:
Hey @ROQUI101, are you using Windows or a NTFS partition ?

ROQUI101, 2024-05-17:
> Hey @ROQUI101, are you using Windows or a NTFS partition ?

Yes, I'm using Windows (Ubuntu with WSL2) and sd images hosted in NTFS partitions

Blindstars, 2024-05-17:
I've seen better results/stability when copying the img onto WSL before shrinking. Seems faster also.

ROQUI101, 2024-05-17:
I have copied the image to /home but I get the same error

[image]

borgesnotes, 2024-05-17:
@ROQUI101, I'm not sure, but I believe that it will not work with an SD card using NTFS. Try formatting the SD card with exFAT instead.

Drewsif, 2024-05-18:
`tune2fs` which is what pishrink uses to shrink the last partition only supports ext2/ext3/ext4 filesystems. It looks like your last partition is exfat/fat32 so thats why it is not working.

It's not a simple thing to add since it needs the ability to detect what type of filesystem is being used and support shrinking and expanding that type. It might be added in the future though. If your last partition is just to store games you could reformat the partition and pishrink will work.

closed 2024-05-18 by Drewsif

ROQUI101, 2024-05-19:
Totally understood. Thanks for the clarification Drewsif.