Drewsif/PiShrink #278: shrinkPi for Sonoma

ruediheimlicher, 2024-05-31:
Hi
I wanted to use PiShrink on my MBP with macos sonoma
I got an error:
install -d /usr/local/bin
install -m 755 e2fsck /usr/local/bin
install -m 755 resize2fs /usr/local/bin
install: resize2fs: No such file or directory

Are you planning an update to use under Macos Sonoma? This would be very helpful.
Regards
Ruedi Heimlicher

borgesnotes, 2024-05-31:
Hey @ruediheimlicher you can use [PiShrink dockerized](https://github.com/borgesnotes/pishrink-docker) | [Docker Hub PiShrink](https://hub.docker.com/r/borgesnotes/pishrink):

1. [Install Docker Desktop on Mac](https://docs.docker.com/docker-for-mac/install/)

2. Make a copy of a Raspberry Pi SD card that you want to shrink ([see instructions here](https://github.com/borgesnotes/raspberry-pi/blob/master/setup/clone-sd-card.md)).

3. Using the Terminal, access the directory containing the Raspberry Pi image:

```bash
cd ~/Directory-with-RPi-image
```

4. Run PiShrink dockerized:

```bash
docker run --privileged=true --rm \
    --volume $(pwd):/workdir \
    borgesnotes/pishrink \
    pishrink -Zv IMAGE_NAME.img NEW-IMAGE_NAME.img
```

Drewsif, 2024-06-04:
There is currently no official support for MacOS, nor will there be in the future as I can't test it. Follow @borgesnotes docker instructions and that should work for you.

closed 2024-06-04 by Drewsif