Drewsif/PiShrink #284: Can't enable auto login after shrink

orbian, 2024-09-29:
I used the shrink script on my image file I created with Win32DiskImager. I flashed a new card with the shrunk image but now I can't enable auto login for the desktop. I enabled it in the settings, but it is ignored now and always brings up a login screen when starting the desktop.

Drewsif, 2024-10-04:
That does not sound related to PiShrink? Do you have any more details or logs on the issue that might help track it down?

The-Exterminator, 2024-10-06:
I have only this trouble with bookworm, i need to chance the password afain in the rasp-config before it works

orbian, 2024-10-22:
What does PiShrink change when it makes the image? The original image booted fairly quickly to desktop. The shrunk image seems to be waiting for something on boot and the auto-login times out and rolls into a login screen for the desktop. I thought it might be related to the disk space expansion the I believe PiShrink turns on. But that seem to have completed successfully.

Drewsif, 2024-10-23:
The expansion happens on the first boot and should trigger a reboot. Does it work as excepted with the auto expansion turned off? I'm not sure how auto login works under the hood well enough to know if it could break something, but I'll try and see if I can replicate the issue.

Geektoolkit, 2025-02-05:
I hit this when trying to use pishrink.  I tried the -s option and it didn't help.  

I'm running pishrink on a pi5 under rasbpian bookworm.

The image I'm shrinking is a Wayland based Bookworm Raspbian image, 64 bit. 

Autologin is enabled before shrinking, however after shrinking and writing the image I am presented with the autologin prompt.