Drewsif/PiShrink #292: Still problems with hard disk filling up completely with file /PiShrink_zero_file

SidewinderGER, 2024-10-22:
`version="v24.10.10"`

First of all, thanks for this great tool!

The recent bugfix for #288 did not work as intended.
In certain situations, PiShrink still fills up the entire machine's space with a file called `/PiShrink_zero_file` containing of all zeroes.

The issue is in line 353ff. The following code got recently added there:
```
  if [ -n "$mount_dir" ]; then
    mountdir=$(mktemp -d)
  fi
```
The purpose of this code is to set `$mountdir` to a temporary path in case it does not have a value assigned already. Unfortunately, there are two issues here:
- The name of the variable being tested is misspelled with an extra underscore, leading to the condition never being true (except if someone set `$mount_dir` before running the script).
- The test operator `-n` (test for non-empty string) is inappropriate, it should have been `-z` (test for empty string).

Hence, the code should rather be:
```
  if [ -z "$mountdir" ]; then
    mountdir=$(mktemp -d)
  fi
```

After applying this fix in my working copy, the script works as expected.

SidewinderGER, 2024-10-22:
I just realized there is another issue with this piece of code.
Immediately following the code block mentioned above there is the following line:
```
rc=$?
```
Before the fix to #288 , this line always captured the return code from the `resize2fs` command. However, with the introduction of the code block for setting `$mountdir` on demand, it may capture the return code of the `mktemp` command, which is most probably not intended. The line `rc=$?` should be moved up above the `if` block.

Drewsif, 2024-10-23:
Wow, what an almost comical set of oversights on my part. Thank you so much for the detailed report and fixes, I'll get them address ASAP.

To anyone who reads this remember to never code exhausted 😅

commit 6b73041a (2024-10-23): Fix moundir testing

Closes #292
closed 2024-10-23 by Drewsif