Drewsif/PiShrink #294: V24.10.23 - won't work on Pi 32bit Bookworm ?

joneyes, 2024-11-03:
I've been scratching my head on this for a few hours now. 
I have a number of Pi's running various editions of Raspbian, most of them use a pibackup.sh and that launches pishrink.sh
In order to try and unravel the problem, I have broken down the two scripts and run them independently.
Using 2 Pis as examples, one running Bookworm, the other Buster. Both mount a network drive and can create .imgs there.

The Pi that is running Buster quite happily shrinks the image (in this example compressing with Gzip too.):

__pi@Coop:~ $ sudo pishrink.sh -advz /media/backup/pi3btest/2024-11-03.img /media/backup/pi3btest/test4.img
pishrink.sh: Creating log file /home/pi/pishrink.log
PiShrink v24.10.23 - https://github.com/Drewsif/PiShrink_

__pishrink.sh: Copying /media/backup/pi3btest/2024-11-03.img to /media/backup/pi3btest/test4.img..._
_pishrink.sh: Gathering data_
_Creating new /etc/rc.local_
_pishrink.sh: Checking filesystem
rootfs: 117373/904960 files (0.2% non-contiguous), 932718/3652608 blocks_
_resize2fs 1.44.5 (15-Dec-2018)
pishrink.sh: Shrinking filesystem
resize2fs 1.44.5 (15-Dec-2018)
Resizing the filesystem on /dev/loop0 to 934619 (4k) blocks._
_Begin pass 2 (max = 115452)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 112)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 13854)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 934619 (4k) blocks long._

_pishrink.sh: Zeroing any free space left
pishrink.sh: Zeroed 157M
pishrink.sh: Shrinking partition
pishrink.sh: Truncating image
pishrink.sh: Using pigz on the shrunk image
/media/backup/pi3btest/test4.img to /media/backup/pi3btest/test4.img.gz
pishrink.sh: Shrunk /media/backup/pi3btest/test4.img.gz from 15G to 1.2G_

However, the 32 bit Bookworm version hangs at the gathering data point and doesn't move on:

_pi@pi3btest:~ $ sudo pishrink.sh -v /media/backup/pi3btest/2024-11-03.img /media/backup/pi3btest/test3.img
PiShrink v24.10.23 - https://github.com/Drewsif/PiShrink_

_pishrink.sh: Copying /media/backup/pi3btest/2024-11-03.img to /media/backup/pi3btest/test3.img...
pishrink.sh: Gathering data_

There's nothing more added to the log file either if i use the -d switch.
The install was completed using the text from your Linux Installation instructions.
All the prerequsites (parted gzip pigz xz-utils udev e2fsprogs) are all listed in their own folders when running a sudo which:
eg: 
_pi@pi3btest:~ $ sudo which parted
/usr/sbin/parted_

It feels like an issue creating the new /etc/rc.local?
There's certainly no .bak verison of it on the bookworm unit
Bookworm:
[image]
Buster:
[image]
Note that this is not a recent file creation?

I'm at a loss now - do you have any suggestions?

joneyes, 2024-11-03:
PS - I've since manually created rc.local.bak - that didn't make any differenct.

Drewsif, 2024-11-03:
Can you attach your logs? Im guessing one of the external utilities we call is hanging with the network drive. Does it work when the image is local?

joneyes, 2024-11-04:
Thanks for the swift response.
If you can tell me which logs you want (and their locations), I'll attach them.
I'll dig out a usb drive with enough storage and try it local tomorrow

Drewsif, 2024-11-04:
there should be a `pishrink.log` the gets written with the `-d` option

joneyes, 2024-11-04:
So i have run a few tests today.
If i mount a local USB drive and manually copy the image files across. pishrink will work - and pretty quickly.
If i use the same mounted USB drive and specify the USB drive as the final destination of the shrunk file (wit the source on the NAS), it will start to copy, but the file transfer stalls at about 4.1GB (of 15GB).
Pishrink fails to write anything useful to the log when it fails either in the file copy from the NAS to the USB, or shrinking on the NAS.

Noting that the exact same version on Buster writes and works on the same NAS.
Both write as root to the NAS.

[pishrink_nas to nas file copy.log](https://github.com/user-attachments/files/17621340/pishrink_nas.to.nas.file.copy.log)
[pishrink copy to USB.log](https://github.com/user-attachments/files/17621341/pishrink.copy.to.USB.log)
[pishrink_localusb.log](https://github.com/user-attachments/files/17621342/pishrink_localusb.log)

joneyes, 2024-11-05:
Is it possible to spin up a version that will do enhanced logging with a -D switch?

closed 2026-05-10 by Drewsif