Drewsif/PiShrink #296: Failure to recognize partition from Bookworm 64 bit image

mfvargo, 2024-11-19:
when i try to shrink an image from a 16G sd card, it fails at the end.

```bash
parted: invalid token: primary
Error: Expecting a file system type.
pishrink.sh: ERROR occurred in line 401: parted failed with rc 1
```

I installed the latest pishink with

```bash
wget https://raw.githubusercontent.com/Drewsif/PiShrink/master/pishrink.sh
```

installed new raspii64 os to 16G sd card with the pi imager.  Booted it up on the pi and did apt update and upgrade.   

on the pi4

```bash
pi@pi-bookworm:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

on my linux workstation

```bash
(base) mfvargo@mike-ryzen:~/pi-images$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.1 LTS
Release:	24.04
Codename:	noble
```

```bash
sudo dd if=/dev/sda2 of=pi4sound.img bs=1M status=progress
```

after ls -l says:

```bash
-rw-r--r-- 1 mfvargo mfvargo 15390474240 Nov 18 16:29 pi4sound.img
```

Command line for pishrink

```bash
(base) mfvargo@mike-ryzen:~/pi-images$ sudo pishrink.sh -d pi4sound.img 
[sudo] password for mfvargo: 
pishrink.sh: Creating log file /home/mfvargo/pi-images/pishrink.log
PiShrink v24.10.23 - https://github.com/Drewsif/PiShrink

pishrink.sh: Gathering data
pishrink.sh: Checking filesystem
rootfs: 72111/222208 files (0.2% non-contiguous), 701925/908243 blocks
resize2fs 1.47.0 (5-Feb-2023)
pishrink.sh: Shrinking filesystem
resize2fs 1.47.0 (5-Feb-2023)
Resizing the filesystem on /dev/loop4 to 900570 (4k) blocks.
Begin pass 2 (max = 6927)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 28)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop4 is now 900570 (4k) blocks long.

pishrink.sh: Zeroing any free space left
pishrink.sh: Zeroed 760M
pishrink.sh: Shrinking partition
parted: invalid token: primary
Error: Expecting a file system type.
pishrink.sh: ERROR occurred in line 401: parted failed with rc 1
```

pisrhink.log file

```bash
(base) mfvargo@mike-ryzen:~/pi-images$ cat pishrink.log 
PiShrink v24.10.23 - https://github.com/Drewsif/PiShrink

pishrink.sh: Gathering data
Line 326
beforesize: 15G
parted_output: BYT;
/home/mfvargo/pi-images/pi4sound.img:15390474240B:file:512:512:loop::;
1:0B:15390474239B:15390474240B:ext4::;
partnum: 1
partstart: 0
parttype: primary
tune2fs_output: tune2fs 1.47.0 (5-Feb-2023)
Filesystem volume name:   rootfs
Last mounted on:          /media/mfvargo/rootfs
Filesystem UUID:          3c2e15cb-bd7f-4963-ac70-c252251ab054
Filesystem magic number:  0xEF53
Filesystem revision #:    1 (dynamic)
Filesystem features:      has_journal ext_attr resize_inode dir_index filetype needs_recovery extent flex_bg sparse_super large_file dir_nlink extra_isize metadata_csum
Filesystem flags:         unsigned_directory_hash 
Default mount options:    user_xattr acl
Filesystem state:         clean
Errors behavior:          Continue
Filesystem OS type:       Linux
Inode count:              912640
Block count:              3757440
Reserved block count:     187869
Overhead clusters:        74984
Free blocks:              3011937
Free inodes:              840530
First block:              0
Block size:               4096
Fragment size:            4096
Reserved GDT blocks:      131
Blocks per group:         32768
Fragments per group:      32768
Inodes per group:         7936
Inode blocks per group:   496
Flex block group size:    16
Filesystem created:       Tue Nov 12 17:13:27 2024
Last mount time:          Mon Nov 18 14:44:19 2024
Last write time:          Mon Nov 18 14:44:19 2024
Mount count:              8
Maximum mount count:      -1
Last checked:             Tue Nov 12 17:13:27 2024
Check interval:           0 (<none>)
Lifetime writes:          3759 MB
Reserved blocks uid:      0 (user root)
Reserved blocks gid:      0 (group root)
First inode:              11
Inode size:	          256
Required extra isize:     32
Desired extra isize:      32
Journal inode:            8
Default directory hash:   half_md4
Directory Hash Seed:      70e529c9-b1c9-4cba-900f-af5f24981700
Journal backup:           inode blocks
Checksum type:            crc32c
Checksum:                 0x07832ace
currentsize: 3757440
blocksize: 4096
pishrink.sh: An existing /etc/rc.local was not found, autoexpand may fail...
grep: /tmp/tmp.uzjWBM3fRy/etc/rc.local: No such file or directory
Creating new /etc/rc.local
pishrink.sh: Checking filesystem
rootfs: 72111/912640 files (0.2% non-contiguous), 745504/3757440 blocks
resize2fs 1.47.0 (5-Feb-2023)
Line 346
currentsize: 3757440
minsize: 903243
Line 352
extra_space: 2854197
Line 359
minsize: 908243
pishrink.sh: Shrinking filesystem
resize2fs 1.47.0 (5-Feb-2023)
Resizing the filesystem on /dev/loop4 to 908243 (4k) blocks.
Begin pass 2 (max = 84560)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 115)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop4 is now 908243 (4k) blocks long.

pishrink.sh: Zeroing any free space left
pishrink.sh: Zeroed 790M
pishrink.sh: Shrinking partition
Line 390
partnewsize: 3720163328
newpartend: 3720163328
parted: invalid token: primary
Error: Expecting a file system type.
pishrink.sh: ERROR occurred in line 401: parted failed with rc 1
```

mfvargo, 2024-11-19:
I notice in the pishrink.sh

```bash
  parted -s "$img" unit B mkpart "$parttype" "$partstart" "$newpartend"
  rc=$?
  if (( $rc )); then
    error $LINENO "parted failed with rc $rc"
    exit 14
  fi
```

Is it that my partd on my ubuntu 24.04 noble does not recognize the partition from the bookworm pi?  Do I need to run pishrink on the pi running bookworm?

closed 2024-11-19 by mfvargo

Drewsif, 2024-11-19:
Thanks for the update! Glad you found what it was.