Drewsif/PiShrink #309: Question about:   WARNING: PiShrink does not yet support autoexpanding of this type of image

AdrianJSmith, 2025-01-14:
Hi,

the following is a test that I ran on a 16gb sd micro card image:

I received the WARNING: PiShrink does not yet support autoexpanding of this type of image

also when the pishrink was completed; it had only shrunk the image file to 15gb from the original 16gb

is that because the original sd micro card was almost full?

Thank for your help.

Adrian Smith 14/01/2025

------------------------------------------------------
16gb image file before pishrink:

root@hp850:~# ls -lrt rasp-pi-16gb-sd-micro-2025-01-14.img

-rw-r--r-- 1 root root 16088301568 Jan 14 19:59 rasp-pi-16gb-sd-micro-2025-01-14.img
root@hp850:~# 

root@hp850:~# pishrink.sh rasp-pi-16gb-sd-micro-2025-01-14.img 
PiShrink v24.10.23 - https://github.com/Drewsif/PiShrink

pishrink.sh: Gathering data
WARNING: PiShrink does not yet support autoexpanding of this type of image
pishrink.sh: Checking filesystem
root: recovering journal
root: 143884/881280 files (0.1% non-contiguous), 1032884/3538432 blocks
resize2fs 1.45.5 (07-Jan-2020)
pishrink.sh: Shrinking filesystem
resize2fs 1.45.5 (07-Jan-2020)
Resizing the filesystem on /dev/loop11 to 1388765 (4k) blocks.
Begin pass 2 (max = 48830)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 108)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 12847)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop11 is now 1388765 (4k) blocks long.

pishrink.sh: Zeroing any free space left
pishrink.sh: Zeroed 1.5G
pishrink.sh: Shrinking partition
pishrink.sh: Truncating image
pishrink.sh: Shrunk rasp-pi-16gb-sd-micro-2025-01-14.img from 15G to 15G
root@hp850:~#

16gb image file after pishrink:

root@hp850:~# ll rasp-pi-16gb-sd-micro-2025-01-14.img 
-rw-r--r-- 1 root root 15768485888 Jan 14 20:27 rasp-pi-16gb-sd-micro-2025-01-14.img
root@hp850:~# 
------------------------------------------------------

closed 2025-01-17 by Drewsif

AdrianJSmith, 2025-01-18:
Hi,

Thank you for your answer; as I said that was just a test that i did on an old raspberry pi 16G micro sd card with Raspbian GNU/Linux 8 (jessie).

I successfully used pishrink to shrink a Raspbian GNU/Linux 11 (bullseye) 64GB micro sd card down to 5.7GB; and then wrote the shrunken image back to a 16GB micro sd card.

It then booted perfectly in a raspberry pi model 3.

Thanks again for a very useful tool.

Kind regards, Adrian Smith Upper Hutt New Zealand

Drewsif, 2025-01-18:
Glad to hear it's working well for you! Cheers 😄