Drewsif/PiShrink #322: Always getting 40GB images after shrinking about 8GB of data.

peregrines2, 2025-05-09:
Hi!

I am trying to shrink multiple Micro SD cards containing Victron Venus OS images (4 partitions each system). Data size is only about 8GB
Unfortunately I am getting 40GB images - whatever I try :(
I am using Ubuntu!
Any hints what I am doing wrong?
Thank You!

[image]

Drewsif, 2025-05-10:
That size is outside of PiShrink's control. Something you could try doing is running [e4defrag](https://man7.org/linux/man-pages/man8/e4defrag.8.html) and see if after that you can shrink the partition more. Try it on the live system and if it lets you shrink the image more after I'll add it into the script!

Drewsif, 2026-05-10:
This is an automated cleanup message. 

This issue is being closed because PiShrink is designed to only shrink the last partition of an image. Victron Venus OS uses a 4-partition layout, meaning the inability to shrink the earlier partitions is a documented limitation of the script, rather than a bug.

If you believe this issue was closed in error, please feel free to reopen it with additional details!