Drewsif/PiShrink #325: Pishrink creates strange partitions

wavoigt, 2025-05-25:
I shrinked my HAOS SSD (120Gb) to 9 Gb.
The .img i made with WIN32DiskImager.
Now, after burning on a new 32Gb SD the SD has 9 (!) partitions:

Original SSD:
[image]
Pishrinked SD card:
[image]

I even cannot find the hassos-boot partition on SD.
So, anyway the SD is working on Raspi.
Very strange.

wavoigt, 2025-06-08:
... no comments?

Drewsif, 2025-06-10:
Been busy IRL so just now seeing this. It looks like your original SSD had 119GB unassigned? So it looks like Windows is confused or there was something there or you had a disk problem. I'm not familiar with HAOS's partition layout but this is very unlikely a result of PiShrink as we only change the last partition of an image. I'd try taking an image on a Linux machine or check the original drive in Linux to see if it's partition table looks different.